xyz-tools/gcode-preview #5: [Questions] Octoprint, Rotation by Keypress/Mouse

OllisGit, 2019-05-07:
Hi @remcoder,
I like this project, because the sourcecode looks really clean and the "shaded output" looks nice.

Did you think about an Octoprint integration? I would be realy nice to the the 3d model instead of the flat plane.
What about rotation on a key press or mouse-movement?
Or superduper sweet would be an orientation-box like in Autodesk-Fusion or like in this JavaScript GCode-Viewer: http://nraynaud.github.io/webgcode/

What do you think?
BR
Olli

remcoder, 2019-05-08:
I should also add that the code is open source so you're free to use it. I would just _really_ appreciate it if you give me credit if you decide to publish it.

However, if you give me some time to dust the project off (I just found some old not-yet-commited changes for instance) I will give it a try to roll a proper plugin myself over the next 2 months or so.

OllisGit, 2019-05-10:
Thanks for your fast and detailed answer!

I was thinking about a simple 3d viewer-plugin beside the current gcode-view. So, no tracking, no simulation like you mentioned...but if it is possible with less effort, why not ;-)

I looked into your thumbnail-plugin (didn't worked, doesn't matter) and I thought is it really a good idea to generate the thumbnails on the client side in "realtime"? I mean the file-manager lists a lot of files and some of the files maybe really big (>10MB). The performance can not be good.
A better aproach could be to render the thumbnails on the server as an image, store this information and add an img-tag to the web-ui....but I have no idea how to render the gcode on the server side.
Or even better: Click on the thumbnail in the file-manager, a popup appears and you are able to rotate the object and then take a screenshot.
After the print is done, the attached camera take also a screenshot and stores this to the filemanager-ui.
But maybe not in the orig. filemanger, instead something like this: https://plugins.octoprint.org/plugins/filemanager/
Now you have a "target-actual comparison"....but I am dreaming ;-)

I googled about "three.js" and "gcode" and found this: https://threejs.org/examples/webgl_loader_gcode.html
It's fast, looks ugly, but maybe it is possible to add your "shader-approach" from black->white!
I think it's slicer independent. Btw. I only use cura.

At the moment I am only thinking about  new usefull Octoprint plugins. I have several on my list, 3d viewer is only one...and if i use your solution of cource I will mention you ;-)

Bye
Olli

remcoder, 2019-05-26:
Hey Olli,

I just wanted to follow-up on this: In the mean time I've created a proper javascript bundle of the gcode previewer to make it easy to the use the code as a library. I've also fixed and improved the thumbs plugin for Octoprint. Neither is finished as I'm still working out some kinks but I've made some progress. In time I'd like to publish the library on npm and the plugin in the OctoPrint repo. And I want to refactor the thumbs plugin to use the library version under the hood. 

As for your our original discussion:
 - scope: I'm turning this repo into a library only
 - for now, this lib will not use webgl for rendering. I kinda like the quick 'n dirty method I have now and it suites my needs ;-)
 - I still think a 3d preview in a tab is good idea but that should be a separate thing. 
 - at this point I'm not planning to create a new plugin myself but I'm willing to help out if anybody else wants to go for it.  (I'm also toying with the idea of using the preview rendering as better progressbar)

So for now I'll focus on getting the library in shape so it can be published and possibly re-used for other plugins. I'll update here whenever there's progress. Maybe next weekend.

OllisGit, 2019-05-29:
Good to know!
And I saw that you also created an issue for the error in the Octoprint-Plugin
```
Cannot read property 'zones' of undefined
    at GCodePreview.getOuterBounds (gcode-preview.js:168)
```
I can't wait to test your plugin/lib.

C Ya,
Olli

PR #7 (2019-11-23): Feature/webgl
This PR adds a WebGL rendered version of the preview which has controls for pan/zoom/rotate. 

[image]

Other changes that went into this PR:
- cleanup demo page
- toggle rendering of travel 
- toggle rendering of extrusion
- use live-reload to build the demo on the fly

This fixes #5 and #6

[demo](https://gcode-preview.web.app/)
closed 2019-11-23 by remcoder