xyz-tools/gcode-preview #8: Speed up

nepeee, 2019-11-25:
Can you add an option to skip every x layers to speed up the loading? (maybe progressive loading)
I'm working on an android based OctoPrint like print software(also lcd replacement for a printer) with web interface and camera preview. I'd like to add some preview option for the web interface and this project looks promising.

remcoder, 2019-11-27:
@nepeee skipping layers can help rendering performance of large models but the loading time is dominated by parsing the text. I'm using `.split('\n')` to separate the lines and I haven't yet found a way to speed that up but I'm open to suggestions.

nepeee, 2019-11-28:
I measured the execution times of the parsing steps with a 47MB gcode file in firefox.
The split('\n').filter(l => l.length>0) takes 100ms of the total 3063ms so its not a big problem.
The big problem is the MoveCommand instantiation, i replaced it with a std object in the parseCommand method:

```
{
  gcode: gcode,
  params: params,
  comment: comment
}
```

and removed the unnecessary "instance of MoveCommand" part in the groupIntoLayers method.

With this modification i reduced the parsing time from 3063ms to 2108ms!
I also tested it in chrome(much slower):
The split \n part is 300ms. With the MoveCommand modification the difference is not that big 3069ms to 2701ms :(

After that i also done the same thing with the Layer class but this somehow increased the execution time with about 100ms in both ff and chrome. Based on this i removed the inheritance from the MoveCommand:
```
class MoveCommand {
        constructor(gcode, params, comment) {
            this.gcode = gcode;
            this.params = params;
            this.comment = comment;
        }
}
```

Result: same speed as with the std object.

After these i went crazy and optimized the s* of the hole thing:
```
const lcnt = lines.length;
const cmtSplit = /;+/g; //~290ms 2335chr
const cmdSplit = / +/g; //~40ms 2625chr
const firstLayerMaxZ = 2;

const layers = [];
var maxZ = 0;
var currentLayer;

for (var i = 0; i < lcnt; i++) {
    const splitted = lines[i].split(cmtSplit); //.trim(); saves 100msec in chrome ~0 in ff
    const cmd = splitted[0];
    const comment = splitted[1];
    const parts = cmd.split(cmdSplit);
    const gcode = parts[0].toLowerCase();

    switch(gcode) {
      case 'g0':
      case 'g1':
        //Not much spped improovement from the parseMove version maybe 10ms
        const params = {};
        const plen = parts.length;
        for (var j=0; j<plen; j++) {
            const cur = parts[j];
            const key = cur.charAt(0).toLowerCase();
            switch(key) {
                case 'x':
                case 'y':
                case 'z':
                case 'e':
                params[ key ] = parseFloat(cur.slice(1));
            }
        }

        var cmd2 = new MoveCommand(gcode, params, comment);

        if (params.z && params.z > maxZ && (maxZ != 0 || params.z < firstLayerMaxZ)) {
          maxZ = params.z;
          currentLayer = new Layer(layers.length, [cmd2]);
          layers.push(currentLayer);
          continue;
        }

        if (currentLayer)
            currentLayer.commands.push(cmd2);
    }
}
```

Final results:
Firefox: 3063ms to 1979ms
Chrome: 3069ms to 2336ms (the diff is the slower line splitting by chrome ~300ms vs 100ms)
I will try next to optimize somehow on the line splitting.

remcoder, 2019-11-29:
Wow nice work! I like what you're doing :-)
Also, I didn't really understand the 'progressive loading' part but now I do. Basically I could parse the gcode in chunks and update the canvas in between. This should improve UX even if it doesn't actually make parsing faster. This is something I'm probably going to look at at some point.

remcoder, 2019-12-08:
Hi @nepeee, I just wanted to mention that I'm [working on incremental parsing & rendering](https://github.com/remcoder/gcode-preview/tree/feature/incremental-parsing). This enables progressive rendering, among other things. 

This also means that some of the optimisations you've shown no longer apply b/c the code changed in the mean time (sorry for that). 

See the updated demo in that branch to see progressive loading in action. Please let me know what you think.

nepeee, 2019-12-09:
Can you upload an unminified version of the gcode-preview.js? I don't know how to work with the ts stuff. Previously i used some online tool to unminify the js but the obfuscation made it hard to read/work with the code.

remcoder, 2019-12-09:
You can disable the 'terser' plugin by commenting it out in `rollup.config.js`:
[image]

To build once:
`yarn run build`

To run the (auto-reloading) demo:
`yarn run demo`

remcoder, 2020-06-16:
Progressive loading is implemented in #13

closed 2020-06-16 by remcoder