xyz-tools/gcode-preview #22: Reverse render

anupam-tiwari, 2021-06-15:
Hey there, is there a possibility to reverse render as in reverse the gcode parse function to remove that part of rendering while keeping the original render?

remcoder, 2021-06-15:
Hi there, I'm not sure I understand. Do you want to process the gcode
before rendering? Could you elaborate on your usecase?

anupam-tiwari, 2021-06-22:
sure, so what I want to do is take a gcode file and renders the model , which is does fantastically. I wanted to add a reverse function as in when I press the button instead of moving forward and rendering, it will go backward and remove gcode from the rendered model

remcoder, 2021-06-24:
You can reverse the animation by first parsing and displaying the gcode, and then lowering `endLayer` in a setTimeout/setInterval loop. 

Something like:
```
 import * as GCodePreview from "gcode-preview";
  
  const preview = new GCodePreview.WebGLPreview({
      targetId: 'gcode-preview',
  });
  
  preview.processGCode(gcode);
  preview.render();
  let maxLayer = preview.layers.length + 1
  setInterval(()=> { 
      preview.endLayer = --maxLayer;
      preview.render();
  }, 100);
```

anupam-tiwari, 2021-06-28:
this is perfect, can it be done using  lines instead of layers?

remcoder, 2021-06-29:
There is no param like maxLine or anything so no.

Another way would be to call preview.processGCode and parse a whole new
gcode fragment. Then every time you can make the gcode fragment shorter.
And in between you should then call preview.clear. This would not really
perform well though so YMMV.

Good luck!

remcoder, 2021-07-15:
Hi @anupam-tiwari, nice work. I can step through the gcode forwards and backwards. Really neat! 

btw, I did notice a glitch I think. Here is see double lines after going backwards & forwards
[image]

Oh and it seems the forwards/backwards buttons seem unresponsive after clicking 'load'. What does 'load' do. exactly?

I think actually that I should update gcode-preview to make it easier to build such an interface, what do you think? What would be needed?

Also, for building interfaces on top of gcode-preview, you may be interested in my [Vue.js example](https://github.com/remcoder/gcode-preview/tree/develop/vue-demo) if you want use [Vue](https://vuejs.org/).

Anyway I appreciate you reported back with your results, thank you. Good luck taking it further if that's your intention. Feel free to ask any questions about gocde-preview again. :-)

closed 2021-07-29 by remcoder