xyz-tools/gcode-preview #23: Can not Test gcode Preview with large file size

oueslati00, 2021-08-31:
I test the demo project with a file size more than 100MegaByte.
the browser need more than 3 GegaByte as memory (RAM) that make my pc more slow until crash
there is solution or this is a normal behavor ?

remcoder, 2021-09-02:
Hi! There is no easy solution for this. The full gcode needs to be loaded
in RAM.

To get around this, I think the gcode needs to be streamed from the server.
But then when some option change, say color, and the gcode needed to be
procent again, it'll need to download the goode again too.

But 3GB RAM for 100MB is a lot. Like 30x times. That's terribly
inefficient. I've actually never looked into that much. I've always written
it up to strings being stored inefficiently in javascript.

If I have time I could try to pinpoint the cause but I can't make any
promises.

If you or anyone else is up to investigate it, please share your results.

remcoder, 2021-09-12:
@oueslati00 I just ran a test with a much larger [chunk size](https://github.com/remcoder/gcode-preview/blob/develop/demo/demo.js#L236) and it reduces memory usage a lot. Maybe you can test your case as well.

The progressive rendering as I've implemented it doesn't really manage the Threejs model very well. It just clears the scene and starts rendering from the start again. I need to look into that.

remcoder, 2021-09-12:
Apparently, I need to properly `dispose()` to [free unused geometries](https://threejs.org/docs/#manual/en/introduction/How-to-dispose-of-objects). I see the same improvement now locally regardless of chunk size.

remcoder, 2021-09-12:
@oueslati00  Should now be improved in 2.4.0. Please let me know what you think.

closed 2021-09-12 by remcoder