xyz-tools/gcode-preview #39: Module '"gcode-preview"' has no exported member 'GCodePreview'.

Zeng95, 2022-10-04:
[image]

remcoder, 2022-10-04:
Which version are you using? 

Try `import * as GCodePreview from 'gcode-preview'` for 2.8.x

In develop I'm working on 3.0 which breaks things

Zeng95, 2022-10-04:
> Which version are you using?
> 
> Try `import * as GCodePreview from 'gcode-preview'` for 2.8.x
> 
> In develop I'm working on 3.0 which breaks things

2.8.1 -> the latest one

Zeng95, 2022-10-04:
[image]
[image]

remcoder, 2022-10-04:
I find I have to reload VS Code before it sees my changes. Maybe that works for you.

I tried a fresh project but I cannot reproduce the type error. Does it build?

Maybe I can help of you create a repo with the issue, if possible.

remcoder, 2022-10-04:
Oh I see the problem. You should be using `.init`

```
import * as GCodePreview from 'gcode-preview'

GCodePreview.init({
   canvas: CanvasRef.current as HTMLCanvasElement
})
```

Zeng95, 2022-10-05:
> Oh I see the problem. You should be using `.init`
> 
> ```
> import * as GCodePreview from 'gcode-preview'
> 
> GCodePreview.init({
>    canvas: CanvasRef.current as HTMLCanvasElement
> })
> ```

Thank you! I'll try it our and let you know if it works

Zeng95, 2022-10-05:
> Oh I see the problem. You should be using `.init`
> 
> ```
> import * as GCodePreview from 'gcode-preview'
> 
> GCodePreview.init({
>    canvas: CanvasRef.current as HTMLCanvasElement
> })
> ```

It works! Awesome! THANK YOU

closed 2022-10-05 by Zeng95

remcoder, 2022-10-05:
Happy to hear!