xyz-tools/gcode-preview #46: Is it possible to add heatmaps

Zeng95, 2022-10-10:
Hi @remcoder. I'm wondering if it's possible to have heatmaps since I'm not too familiar with three.js

remcoder, 2022-10-10:
Do you mean coloring the line segments according to nozzle temp?

I could create a method to let you colorize the line segments according to
the current state, nozzle temp, z-height etc.

What's your usecase?

Zeng95, 2022-10-10:
My use case would be to indicate where there is a weakness in a 3d object.

Zeng95, 2022-10-10:
[image]

remcoder, 2022-10-10:
Maybe you can annotate the GCODE?
G1 X2 Y3 E4
; START HIGHLIGHTED AREA
G1 X3 Y4 E5
; END HIGHLIGHTED AREA
G1 X4 Y5 E6

Then the lines between start / end can be rendered in a different color.

Would this work for you?

Zeng95, 2022-10-11:
> Maybe you can annotate the GCODE? G1 X2 Y3 E4 ; START HIGHLIGHTED AREA G1 X3 Y4 E5 ; END HIGHLIGHTED AREA G1 X4 Y5 E6 Then the lines between start / end can be rendered in a different color. Would this work for you?
> […](#)

I can't guarantee that would work for me, but if I can have the method you mentioned before would be more appreciated 🙏🙏

remcoder, 2022-10-11:
Let's take this example:

```
G1 X2 Y3 E4
; START HIGHLIGHTED AREA
G1 X3 Y4 E5
; END HIGHLIGHTED AREA
G1 X4 Y5 E6
```

To recognize the highlighted area this is needed:

- The parser would need to recognize the marker strings and store the line numbers
- The renderer will need to check for gcode sequences that are interrupted (or intersected) by the markers.
- Those gcode sequences need to be split and rendered separately (currently they are drawn as a single long continuous line). (this might impact performance somewhat)
- The highlighted gcodes need to be drawn in a different color. This color is either :

  1. predefined. 
  2. passed as a param to the constructor or
  3. part of the marker in gcode (`START HIGHLIGHTED AREA RED`)

What do you think? 

I guess it all depends on where you get the data about the weakness in the 3d model. Is this coming from a CAD program? User input? Machine vision? 

And in what form you do get this data? Is the weakness area itself a 3d object?

Zeng95, 2022-10-25:
> Let's take this example:
> 
> ```
> G1 X2 Y3 E4
> ; START HIGHLIGHTED AREA
> G1 X3 Y4 E5
> ; END HIGHLIGHTED AREA
> G1 X4 Y5 E6
> ```
> 
> To recognize the highlighted area this is needed:
> 
> * The parser would need to recognize the marker strings and store the line numbers
> * The renderer will need to check for gcode sequences that are interrupted (or intersected) by the markers.
> * Those gcode sequences need to be split and rendered separately (currently they are drawn as a single long continuous line). (this might impact performance somewhat)
> * The highlighted gcodes need to be drawn in a different color. This color is either :
>   
>   1. predefined.
>   2. passed as a param to the constructor or
>   3. part of the marker in gcode (`START HIGHLIGHTED AREA RED`)
> 
> What do you think?
> 
> I guess it all depends on where you get the data about the weakness in the 3d model. Is this coming from a CAD program? User input? Machine vision?
> 
> And in what form you do get this data? Is the weakness area itself a 3d object? An STL?

Thank you very much for the response. I think I'll close the issue for now

closed 2022-10-25 by Zeng95

remcoder, 2022-10-25:
👍🏻