xyz-tools/gcode-preview #55: G2 / G3 commands are ignored?

RickRyan26, 2023-01-13:
I'd love to use this package but I need arcs.
Is there a way to use G2/G3/G02/G03 with gcode-preview?

remcoder, 2023-01-14:
Hi @RickRyan26, G2/G3 arcs are currently not supported. I don't have the time right now to dive into it but I'd love to support this at some point. It has been on my list for a while now. 

I could sure use any kind of help, like a PR, to get started, if you're up for it.  

Otherwise you might look at this other gcode viewer by @Sindarius, which supports G2/G3 arcs and it is overall really good: https://github.com/Sindarius/npm_gcodeviewer

Sindarius, 2023-01-14:
If you look in my source you can adapt doArc@src/utils.js to add support for g2/g3 in your viewer without too much work I bet.

remcoder, 2023-01-14:
Thx @Sindarius! That will be helpful :-)

RickRyan26, 2023-01-15:
I created a fork, tried to add the G2 functionality based on Sindarius' doArc, and thought I had it, but all my arcs come out flat lol was a fun attempt. Thanks for the thoughtful comments above

remcoder, 2023-01-15:
I'll reopen this issue because it is a valid feature request that should get addressed at some point. Also with the example code and your first steps it would be a shame to let that go to waste.

remcoder, 2023-01-15:
@RickRyan26 A new version was pushed to npm just now. Let us know if it works for you.

RickRyan26, 2023-01-15:
@remcoder I scrolled through your changes, awesome work, much more complete than my attempt!

And I just tried your latest version but the following seems to still draw a straight line.
`const gCode = 'G0 X0 Y0 Z0.2\nG2 X42 Y42 R21 E10';`

remcoder, 2023-01-16:
There seems to be an unrelated bug causing the first extrusion to be skipped sometimes. Could you try adding 1 or 2 G0/G1 extrusions before the G2? 

Otherwise it might be in the code that calculates I and J from the R param, which I didn't test well enough yet. 

I will look at the latter first and report back.

Sindarius, 2023-01-16:
G0 X0 Y0 Z0.2
G2 X42 Y42 R21 E10
Based on those two lines of gcode the G2 is an incorrect value.  R21 would not fit the math.  The minimum R value can't be 21
Find the midpoint between the two points
x = 21, y =21
Apply the Pythagorean theorem to find our R to point 42,42 from 21,21
sqrt(21^2 + 21^2)
you'll find the minimum R is around 29.7

RickRyan26, 2023-01-16:
@Sindarius Thank you!! I would've never figured that one out. Foolish of me to be using unvetted code to test with, it just looked so simple.
Sorry for the bad info @remcoder

RickRyan26, 2023-01-16:
@Sindarius, Even tho your solution does create arcs successfully, the mystery continues. I use Mach3 to run gcode for plasma cutting steel on my CNC table. It's slightly different than 3D printing but I'm regexing my gcode with JS so that it works with the gcode preivew format you guys are using.
It seems that Mach3 uses a different equation than you are since all of my arcs show up and work correctly in Mach3 but are flat with your doArc logic. The following is a vetted example.

`gCode = 'G1 X0 Y0 Z0.06\nG2 X2.22 Y6.45 R0.4 E10'`

Sindarius, 2023-01-16:
Is there a g17 g18 or g19 anywhere in your gcode? That specifies what plane the arc should be in. The code currently handles XY plane at the moment if that's what you mean by "flat"

RickRyan26, 2023-01-16:
My code does not use g17, g18, or g19. And my arcs are all on the XY plane, no vertically slopping arcs. By flat i'm mean it functions the same as a straight G1 line, connecting the XY while ignoring the R. The problem can be seen when running this gcode: `G1 X0 Y0 Z0.06\nG2 X2.22 Y6.45 R0.4 E10`

RickRyan26, 2023-01-17:
The way Mach3 is reading this gcode is more like...
Create a circle with radius 0.4.
Have the perimeter of the circle touch both the starting and ending XY coordinates.
Erase the remaining part of the circle/arc not between the 2 coordinates.

Sindarius, 2023-01-17:
I'll have to see if I can find Mach3s implementation. The implementation I have is based on how RRF handles arcs. I tried a couple other online viewers that have g2 and g3 and they also fail to render that code, presumably because there is not a circle with a radius that small that could touch both those points.

RickRyan26, 2023-01-17:
I am using G02 instead of G2, but I don't think that matters.
To be completely transparent, here is a plasma file of a circle and square (each with arc lead-ins) that I have run successfully.
`
G20 (Units: Inches)
F140 (Initialize feed rate)
G53 G90 G40 (Turn on absolute pos without cut compensation)
S900 (Global DTHC ON/OFF is ON)
S510 (Turn off DTHC at the beginning)
S3128 (Preset volts)
S502 (DTHC Delay set to 0.1 seconds)
S701 (Clear S Register)
(M00 , Pause. Check the DTHC Settings)
G00 Z1.5 (Travel height)
X1.82 Y6.05 (Go to start point)	
(Probe for live pierce height)
M900 (Check for z active)
G00 Z0.2 (Start calibrating Z)
G31 Z-11.8 F55 (Start probe touch-off)
G92 Z0 (Save Z axis ref)
G00 Z0.039 (Switch offset Lift)
G92 Z0 (Save Z axis ref again)
G00 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G04 P0.2
G01 Z0.06 F55.0 
S20 (DTHC is on)
(Cut square, tube #0 )
G02 X2.22 Y6.45 R0.4
G01 Y6.45
X2.22
Y5.65
X1.42
Y6.45
X2.22
S10 (DTHC OFF)
M10P1 (Quick Torch off)

M05 (Torch mode off)
G00 Z1.5 (Travel height)
X3.845 Y2 (Go to start point)
(Probe for live pierce height)
M900 (Check for z active)
G00 Z0.2 (Start calibrating Z)
G31 Z-11.8 F55 (Start probe touch-off)
G92 Z0 (Save Z axis ref)
G00 Z0.039 (Switch offset Lift)
G92 Z0 (Save Z axis ref again)
G00 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G04 P0.2
G01 Z0.06 F55.0 
S20 (DTHC is on)
(Cut circle #0, tube #1, side #1 )
G02 X3.59 Y1.745 R0.255 140
X4.1 Y2.255 R0.255
X3.59 Y1.745 R0.255
S10 (DTHC OFF)
M10P1 (Quick Torch off)

M05 (Torch mode off)
G00 Z1.5 (Travel height)
X0 Y0 (Go to origin)
S900 (Global DTHC ON/OFF is ON)
S10 (DTHC is off)
S701 (clear S Register)
M05 M30 (End and reset program)
`

Sindarius, 2023-01-17:
Thanks for sharing the g-code. That helped out quite a bit in figuring out what's going on. 
1. The numbers are in inches I don't know if this viewer supports inches (it is on my todo on my viewer)
2. Because the numbers are in inches the segments are going to come out small since they are set for a min segment size.
3. I am not sure if this viewer supports absolute position since it appears the g-code is completely in absolute (G90).
4. Your last sample (0,0 -> 2.22,6.25)  is not quite right based off this g-code. The arc for the square starts at (1.82, 6.05) not (0,0),  which would allow that arc to be computed properly. (I tweaked my viewer's segment size to verify)
5. The lead-in cut for the circle works but the circle itself is smaller than the minimum radius and fails to render. Given the XY start and end of the circle (3.59,1.745)(4.1,2.255) the minimum radius would be >0.3606 so 0.255 fails to render not sure what Mach3 is doing to allow that to be cut. 

For giggles I changed the segment size for arcs to 0.001 so I could render them out (until I fix my inch conversion) 

[image]

If I change those last 2 G2's to the minimum I get the circle
[image]

Ignore the line off into space that was me moving the cursor.

All in all I am sure @remcoder can implement these items given time, and it helped me to validate some items on my side as well so thanks!

remcoder, 2023-01-17:
I also don't understand how Mach3 wants to draw that circle. I checked with [ncviewer](https://ncviewer.com/), which runs the gcode as-is but it will also not produce a circle. Although it does try to do something. 

[image]

@RickRyan26  Maybe you could re-check your gcode and/or generate some more examples?

RickRyan26, 2023-01-18:
@remcoder @Sindarius 
Here's proof my example works for me. I am using the file as a `.tap` instead of a `.gcode` but I wouldn't think that matters

[image]

Also for more examples.... here's the preprocess JS i'm using to generate holes because Mach3 can't handle gcode with while loops in it.

```
G00 Z${travelHeightReal} (Travel height)
X${nextX} Y${nextY} (Go to start point)
(Cut circle #${hole}, tube #${tube}, side #${side + 1} )
G02 X${ff(nextX - halfHole)} Y${ff(nextY - halfHole)} R${halfHole} ${feedRate}
X${ff(nextX + halfHole)} Y${ff(nextY + halfHole)} R${halfHole}
X${ff(nextX - halfHole)} Y${ff(nextY - halfHole)} R${halfHole}
```

Sindarius, 2023-01-19:
Wild, if you change those other two 2.55's to something like 1 does it render the same or does the circle change? I am wondering if they are fixing the radius internally.

RickRyan26, 2023-01-19:
Wild indeed. Here's what it looks like with the 2nd 0.255 changed to 1 and the 3rd changed to 0.1.

[image]

RickRyan26, 2023-01-19:
Clicking this will download a Mach3 demo you can mess around with if you want.

https://www.machsupport.com/wp-content/uploads/2013/04/Mach3Version3.043.exe

remcoder, 2023-01-19:
So apparently Mach3 will indeed fix a too small radius by stretching it up to the minimum required, in this case ±0.36.

If I change the last code by @RickRyan26 into this I can reproduce the rendered toolpath using ncviewer:
```
G02 X3.59 Y1.745 R0.255 F140
X4.1 Y2.255 R1
X3.59 Y1.745 R0.361
```

[image]

I suggest this behavior should be behind a flag (say 'AllowSmallArcRadius' or maybe some better wording) as I don't think it is the default for 3d printers (which is the primary usecase for this viewer). 

I'll gladly test and merge a PR that fixes this case.

Sindarius, 2023-01-19:
With your AllowSmallArcRadius you'll want to recompute the minimum radius, then you can resolve hSquared and hDivD. You can use the following to get the minimum
`const minR = Math.sqrt(Math.pow(deltaX / 2, 2) + Math.pow(deltaY / 2, 2));`

remcoder, 2023-02-24:
@RickRyan26 it took a while but I've added the above change (without a flag, yolo). Arc segments have a 25x higher resolution in G20 mode. 

fixed in https://github.com/remcoder/gcode-preview/releases/tag/v2.9.2

closed 2023-02-24 by remcoder

RickRyan26, 2023-02-25:
Hey @remcoder,

Haven't gotten to dive into this fully and again I'm using your package in a slightly funky way since I need it for plasma cutting on 2inch steel tube, not 3D printing, but it seems like almost everything that was supposed to be an straight line is now an arc, rather than my diagonal dashes becoming circles. My goal is to have a bunch of circles in rows with some squares at the end.

v2.9.2
[image]

v2.9.0
[image]

Code preview
```G0 (Units: Inches) E10
F112.5 (Initialize feed rate)
G3 G90 G40 (Turn on absolute pos without cut compensation) E10
(S900 :: Global DTHC ON/OFF is ON)
S510 (Turn off DTHC at the beginning)
S3128 (Preset volts)
S502 (DTHC Delay set to 0.1 seconds)
S701 (Clear S Register)
M00 (Pause. Resume now or skip ahead)
()
( All cuts start below this line )
()

(Side #1, Tube #1, Square #1)
G0 Z0.375 (Travel height)
X1.7551 Y2.4 (Go to start point)
(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0
(Square, Tube #0 )
G2 X2.1051 Y2.8 R0.35 E10
G1 Y2.75 E10
G2 X2.1051 E10
G2 Y2.05 E10
G2 X1.4051 E10
G2 Y2.75 E10
G2 X2.1051 E10
M10P1 (Quick Torch off)

M05 (Torch mode off)
(Side #1, Tube #1, Circle #1)
G0 Z0.375 (Travel height)
X1.8301 Y1.25 (Go to start point)
(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0
G2 X1.5711 Y0.991 R0.259 F112.5 E10
G2 X2.0891 Y1.509 R0.259 E10
G2 X1.5711 Y0.991 R0.259 E10
M10P1 (Quick Torch off)

M05 (Torch mode off)
(Side #1, Tube #1, Circle #2)
G0 Z0.375 (Travel height)
X1.8301 Y3.625 (Go to start point)
(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0
G2 X1.5711 Y3.366 R0.259 F112.5 E10
G2 X2.0891 Y3.884 R0.259 E10
G2 X1.5711 Y3.366 R0.259 E10
M10P1 (Quick Torch off)

...
```

remcoder, 2023-02-25:
O wow didn't expect that!

Good to hear you're still using the package so I'll do my best to figure
this out over the weekend.

Could you please add a screenshot from Mach3 so I can see what it is trying
to do?

In the meantime, you can fall back to v2.9.0 I guess? Let me know if you
have trouble with that.

Thanks for your quick feedback!

remcoder, 2023-02-25:
I just noticed a few lines that don't start with a G-code or M-code. That
will not work. But that is not the big issue here..

remcoder, 2023-02-25:
I also don't see a G20 at the top so it will interpret everything as
millimeters. And then the arcs will be too small to be divided in segments.

Can you check if that improves things?

remcoder, 2023-02-25:
If you're using a script to preprocess the gcode I'd like to use/see it, if possible.

remcoder, 2023-02-26:
I fixed a bug where the start of the gcode was skipped causing al sort of problems. This in itself calls for a new version so I'll publish one.

Anyway, I now see one of the 2 circles. The second one is still now working however.

[image]

I'm using this version of your gcode btw. Changing some lines that started with `X123` into `G0 X123`.

```
;(Side #1, Tube #1, Square #1)
G20
G0 Z0.375 (Travel height)
G0 X1.7551 Y2.4 (Go to start point)

;(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
;G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
;G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0

;(Square, Tube #0 )
G2 X2.1051 Y2.8 R0.35 E10
G1 Y2.75 E10
G2 X2.1051 E10
G2 Y2.05 E10
G2 X1.4051 E10
G2 Y2.75 E10
G2 X2.1051 E10
M10P1 (Quick Torch off)

M05 (Torch mode off)
;(Side #1, Tube #1, Circle #1)
G0 Z0.375 (Travel height)
G0 X1.8301 Y1.25 (Go to start point)
;(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0
G2 X1.5711 Y0.991 R0.259 F112.5 E10
G2 X2.0891 Y1.509 R0.259 E10
G2 X1.5711 Y0.991 R0.259 E10
M10P1 (Quick Torch off)

M05 (Torch mode off)
;(Side #1, Tube #1, Circle #2)
G0 Z0.375 (Travel height)
G0 X1.8301 Y3.625 (Go to start point)
;(Probe for live pierce height)
M900 (Check for z active)
G1 Z-12 F55 (Start probe touch-off)
G2 Z0 (Save Z axis ref)
G0 Z0.039 (Switch offset Lift)
G2 Z0 (Save Z axis ref again)
G0 Z0.15 (Pierce height)
M03 (Torch mode on)
M11P1 (Quick torch on)
G4 P0.2 E10
G1 Z0.06 F55.0
G2 X1.5711 Y3.366 R0.259 F112.5 E10
G2 X2.0891 Y3.884 R0.259 E10
G2 X1.5711 Y3.366 R0.259 E10
M10P1 (Quick Torch off)
```

RickRyan26, 2023-02-26:
@remcoder thanks for working on this! Youre awesome. I'm currently having a busy weekend but your screenshot is exactly what I'm looking for, just with all the circles. I'll get you more information tomorrow and will send you my script so that we can get this solved right. Note that the arc leading into the square is slightly off on purpose, just looks like an error.

RickRyan26, 2023-02-28:
Here's everything I got.

So my website dynamicallt creates gcode files like the attached; this one will help me create the parts I need to build a 13ft 9in Driveway Gate.

And here is my script that mutates the gcode into the format for your viewer; it's not gorgeous but it works.

```
let GCodePreview;
const showDownloadModal = (gCode) => {
	setTimeout(async () => {
		const previewCode = gCode
			.split('\n')
			.map((i) => {
				const line = i?.trim?.();
				let result = line;
				if (line?.charAt?.(0) == 'G') {
					result = `G${line.slice(2)}`;
					if (!line?.includes('Z')) {
						result += ' E10';
					}
				}
				if (!line.includes('start') && ['X', 'Y'].includes(line?.charAt?.(0))) {
					result = `G2 ${line} E10`;
				}
				return result;
			})
			.join('\n');
		const canvas = document.querySelector('canvas');
		GCodePreview ??= await import('gcode-preview');
		const preview = new GCodePreview.init({
			canvas
		});
		preview.processGCode(previewCode);
	}, 100);
};
```

[GCode.txt](https://github.com/remcoder/gcode-preview/files/10845317/GCode.txt)

remcoder, 2023-03-02:
Thanks, I'll take a look soon.

remcoder, 2023-03-05:
@RickRyan26 here's an updated version of your conversion script:
https://gist.github.com/remcoder/2c99e809124d377e7e3f205accbcbbb2

I think the following cases should be handled a little better now:
- G00/01/02/G03 is no longer shortened to G00/G01/2/G3 because GCode Preview actually should work with conversion. This also means commands like G20 won't be accidentally mangled
- it 'remembers' te last command that started with a G. If it then encounters a line starting with an X or Y it will prepend that command. 

I guess we're missing 2 circles in the lower left corner but overall it doesn't look bad maybe?

You might also want to leave out the + " E10" trick because the arcs are rendered anyway. If you want to visualize the difference between travel and cutting I'd suggest you open a new issue for that. We can define a command for 'cutting on' and 'cutting off' for instance.

Let me know what you think.

[image]

RickRyan26, 2023-03-07:
Wow this look plenty perfect for now! My team is going to love this. Thanks for correcting my code. I have a super busy start to this week but will get this running as soon as I can and will let you know =)

RickRyan26, 2023-03-10:
It's working great on my end now. This is going to help a lot in the coming months. Thanks again!

Last question, is there a simple way to set the starting zoom? My current plan is to scale up all the numbers equally.

remcoder, 2023-03-11:
For setting the zoom, the `.camera` property is accessible ([docs](https://threejs.org/docs/#api/en/cameras/PerspectiveCamera)). I think you should be able to manually move to a nice position + zoom level and then read out the transformation matrix for later use. Good luck! :-)

btw, I would love the see a picture of the plasma cutter you're working with. Could you share a photo maybe? :-) And what website are you running the gcode preview on?