xyz-tools/gcode-preview #59: fast movements are displayed as extrusion

hildnerb, 2023-04-21:
Hello, super project.
But I have a problem with the display. It seems that the fast movements are displayed as normal extrusion.
I used the same Benchi as in the example code.

Could you help me further?

[image]

```typescript
import { AfterViewInit, Component, ElementRef, ViewChild } from '@angular/core';
import { WebGLPreview } from 'gcode-preview';
import { FileSystemFileEntry, NgxFileDropEntry } from 'ngx-file-drop';

@Component({
  selector: 'app-root',
  templateUrl: './app.component.html',
  styleUrls: ['./app.component.css']
})

export class AppComponent implements AfterViewInit {
  file: any;
  preview: any;
  private context: HTMLCanvasElement;
  @ViewChild('myCanvas', { static: false }) canvas: ElementRef;

  ngAfterViewInit(): any {
    this.context = (this.canvas.nativeElement as HTMLCanvasElement);

    this.preview = new WebGLPreview({
      canvas: this.context,
      //targetId: 'gcode-preview',
      //topLayerColor: new THREE.Color('lime').getHex(),
      //lastSegmentColor: new THREE.Color('red').getHex(),
      buildVolume: { x: 150, y: 150, z: 150, r: 0, i: 0, j: 0 },
      initialCameraPosition: [270, 300, 300],
      lineWidth: 4,
      allowDragNDrop: true
    });

    this.preview.renderExtrusion = true;
    this.preview.renderTravel = false;

    this.preview.render();

  }

  dropped(files: NgxFileDropEntry[]): any {
    for (const droppedFile of files) {
      if (droppedFile.fileEntry.isFile && droppedFile.relativePath !== undefined) {
        const fileEntry = droppedFile.fileEntry as FileSystemFileEntry; // cast droppedFile to FileEntry
        fileEntry.file(async (file: File) => {
          const url = window.URL.createObjectURL(file); // createUrl To The FileEntry
          const linesList = await this.fetchGcode(url); // get List Of Line of the current File
          this.preview.processGCode(linesList);
        });
      }
    }

    this.preview.renderTravel = false;

    //Canvas2Image.saveAsJPEG(gcodePreview.canvas,innerWidth, innerHeight, fileName.innerText.replace('.gcode','.jpg'));
  }

  async fetchGcode(url): Promise<any> {
    const response = await fetch(url);

    if (response.status !== 200) {
      throw new Error(`status code: ${response.status}`);
    }
    const file = await response.text();
    return file.split('\n');
  }

}

```

remcoder, 2023-04-21:
hi @hildnerb,

It must be the `.lineWidth` that you've set. When you set it to any value it will render this way unfortunately. I never got this working properly. So, to fix it just comment out that line. 

I should remove the option altogether I guess. Eventually I want to render the lines as if they have volume. I hope to get around to it at some point.

cheers,
Remco

hildnerb, 2023-04-21:
super, thanks for the quick reply :+1:

closed 2023-04-21 by hildnerb