xyz-tools/gcode-preview #65: Cannot set initial camera position.

IGMontero, 2023-08-21:
Hello! I'm trying to set the initial camera position based on a point I have, but no matter what I do, I cannot change the X and Z position of the initial camera position.

The only way to do it but which doesn't work for me, was to set it in the "change" event listener for the controls. 
But again, this doesn't work because I want the user to be able to change that position:
[image]

Tried multiple things, like running a timeout that sets the position every n seconds, but again, not working.

Any ideas on how to resolve this?

These are some hardcoded parameters that I'm using for example. But they are not working:
[image]

The only value that actually works is the Y (3000).

If you have any ideas please let me know!

Thanks

IGMontero, 2023-08-21:
Some additional info, when I run this code, I see that in the console it's displayed "units set to inches". 
Could this be related to the issue?

Thanks!

remcoder, 2023-08-21:
Here's a little more context:
 - the way the initial position is set is via this code: `this.camera.position.fromArray(this.initialCameraPosition);`
 - It is a [PerspectiveCamera](https://threejs.org/docs/#api/en/cameras/PerspectiveCamera) from THREE.js
 - which has a base class [Camera](https://threejs.org/docs/#api/en/cameras/Camera)
 - which in turn is based on [Object3D](https://threejs.org/docs/#api/en/core/Object3D)
 - which has a `position` property that is a [Vector3D](https://threejs.org/docs/#api/en/math/Vector3)
 - ...which finally has a `fromArray` method
 
 So it's all THREE.js really ;-)

IGMontero, 2023-08-21:
Hey!
I just realized it didn't work because buildVolume was set to 0, 0, 0. If I set the buildVolume to a value that contains the figure, then it kind of works. 

Is there a way to set the initial camera position without rendering the build volume grid?

IGMontero, 2023-08-21:
Another thing -- looks like if I put a bounding box of x: 1500, y: 1500 and then an initial position of x: 0, y: 0, the preview starts at the middle of the bounding box instead of the origin 0, 0.

Is there anything I can do to change the initial position of the camera in this case?

Thank you a lot for your help by the way! Trying to understand the functionality and limitations of the library

remcoder, 2023-08-25:
To render without a buildVolume grid you can set de buildVolume of `undefined`. But the buildVolume is used for centering also. 

To understand how this works underneath atm [refer to this code](https://github.com/remcoder/gcode-preview/blob/develop/src/webgl-preview.ts#L246-L251).

Without a buildVolume, the only thing to center on is the gcode itself but slicers often add start/end gcode which makes it hard to center on the model by looking at the gcode alone.

Maybe it would be helpful if the buildVolume can be defined (or read from the gcode) but still not rendered. Having a numerically defined buildVolume will at least allow for proper centering.

I could add an configuration option for that if you want.

remcoder, 2023-09-10:
@IGMontero just checking if your question was answered and if we can close this isssue?

IGMontero, 2023-09-10:
Yes, thank you very much! 🙌

closed 2023-09-10 by remcoder