xyz-tools/gcode-preview #70: G2 G3 offsets

VladimirSmar, 2023-09-07:
[image]
Idk if its specific to my case, but here laser machine performing offset calculations from 0.0 point, in short I and J are coordinates for arc center. Because of that lib cant really draw those arcs properly. From what i saw in code, center is calculated from last point coordinates + I J. Is there's a way to fix it? Maybe set a point from which offsets will be set? Or this library cant really do it?
(white part of image is a mesh that im drawing myself going through same gcode file)
Also its kinda awkward, but im forced to substract end point coordinates from offsets, or otherwise its even worse, like circle which is supposed to be like 10R becomes like 100R because of offsets calculated from last point...

remcoder, 2023-09-10:
Hi @VladimirSmar, absolute arc offsets are not supported as they are more of an exception (see [cnccookbook](https://www.cnccookbook.com/cnc-g-code-arc-circle-g02-g03/)) but if you are willing to create a PR for it, I'll review it and help you to get it merged.

A global option could be a way to toggle the behaviour between absolute vs relative arc offsets, something like: `absoluteArcOffsets = true`.

remcoder, 2023-09-10:
Just to be on the same page, I had to modify the gcode above as follows to make it work at all (turning on 'travel'):

```
G30 X0 Y0
G31 X24.0 Y24.0
G45 R0
G71
G90
G0
G41 X0.637 Y2.372
G1 X2. Y7.183
G1 Y16
G2 X8 Y22 I8 J16
G1 X16
G2 X22 Y16 I16 J16
G1 Y9
G2 X20 Y7 I20 J9
G1 X19
G2 X17 Y9 I19 J9
G1 Y15
G3 X15 Y17 I15 J15
G1 X9
G3 X7 Y15 I9 J15
G1 Y2
G1 X2
G1 Y6.533
G0 G40
```

which results in:
[image]

Is that what you are getting?

remcoder, 2023-09-10:
I see you're also using a non-zero cutter compensation via `G41` (line 7 in my snippet: `G41 X0.637 Y2.372`). This is also not yet supported and I again I'd welcome PR's for that. 

To explain: I create the lib for use for 3D printing but since them _some_ support for CNC's has been added. The thing is however that there are many gcode variants for CNC's and that makes testing for all use cases almost impossible. But if you really want to use my lib and make a PR,  I'll help you to get it merged. 

btw, what machine type are you using?

VladimirSmar, 2023-09-19:
Im not really sure which type im using tbh) its just a laser cut machine, but i dont know any specific about it. Also im trying to draw those details w/o those huge circle (movements) =) Thats why im kinda forced to recalculate angles and offsets.

VladimirSmar, 2023-09-19:
Also, as you can see, even with those travel codes straight lines are kinda scuffed and cut a little. I solved this problem exacly with angles.

remcoder, 2024-01-12:
Closing this for now due to not enough info atm.

Anybody is free to reopen this ticket and I that case I'd like a clear test case we can start with.

closed 2024-01-12 by remcoder