xyz-tools/gcode-preview #71: Vase Mode Previewing

FlatTony, 2023-09-29:
I am seeing lots of lag and unresponsiveness when loading files that are sliced in vase mode (constant change in Z). The files are less than 15Mb so it should be able to handle it fine from the point density. I am attaching a file that will not render. The bed size is 635,635,635 for reference. If this can be resolved, it is much appreciated. I will be working on it as well and report back with any findings. 

Cheers.

[FlowerPot.zip](https://github.com/remcoder/gcode-preview/files/12771696/FlowerPot.zip)

remcoder, 2023-10-30:
Hi @FlatTony, ah yes this must be caused by the fact that a new layer is created in memory for each unique z height. Then gcode commands are grouped by layer. This offers some benefits like easy rendering up to a certain layer but it als also has some drawback.. and vase mode especially suffers. 

I think in the end grouping by layer wasn't the best idea but it's going to take rewriting parts of the parser / renderer to really fix this. For now this isn't something I'm going to spend time on myself. 

If you feel like having a go a PR yourself, I'm more than happy to help your out where I can. 

cheers.

remcoder, 2023-11-21:
fixed in #82

closed 2023-11-21 by remcoder

remcoder, 2023-11-21:
@FlatTony this was fixed in v2.11 which was just released. The FlowerPot loads fine now ;-)