xyz-tools/gcode-preview #80: Possibility to change color of model

ivan-hryshko, 2023-10-30:
Give the opportunity to change the color of the model, because now it is gray by default. Is it possible to change this with parameter passing?

remcoder, 2023-10-30:
Hi @ivan-hryshko , setting the extrusion color is not yet possible.

I can add it for you.

Would you like something like this?

yellow, 50% saturation, , brightness [0, 80]
[image]

red, 50% saturation, brightness [0, 80]
[image]

purple, 100% saturation, brightness [20, 80]
[image]

I'm thinking about a param to override the Color, which will determine the hue and the saturation. The brightness will still be based on the layer height. How is that?

Also see the current [code for the gradient](https://github.com/remcoder/gcode-preview/blob/develop/src/webgl-preview.ts#L221-L224).

ivan-hryshko, 2023-10-30:
@remcoder Looks cool, I like it)
The transfer of hue and saturation will be enough for my needs.
Although if you add the ability to set the brightness [20, 80] as you have shown, that would be an added plus as well.
Please let me know when this feature will be added

closed 2023-10-30 by remcoder

remcoder, 2023-10-31:
Just released 2.10 with these changes

ivan-hryshko, 2023-11-26:
@remcoder 
Thank you very much. It looks great.
But is it possible to pass a custom range for brightness as you suggested above?
Because I have a problem with displaying black, gray and white.
Is it possible to set for:
black color for example brightness in the range [0-30]%
for gray [30-70]%
And for white [70-100]%

commit eb3adff1 (2023-10-30): Cleanup after adding extrusionColor. Fixes #80
PR #81 (2023-10-30): Allow extrusion color, etc to be changed
This PR fixes #80 by adding and option for specifying the extrusion color. The same gradient will be applied but the hue and saturation will be kept from the specified color.

The extrusion color can be passed to the constructor or via an accessor. The value can be either a css string (ex '#00ff00', 'hotpink'), a number (ex 0xff00ff) or a Threejs Color instance.

The same will be done for the background color (thx to @raulodev for suggesting the code for this) and the color of the top layer color, the last segment color and the travel color.

 - [x] ext [... cut]