xyz-tools/gcode-preview #83: [demo] sidebar doesn't update

remcoder, 2023-11-21:
When uploading a new gcode file, these are not updated:
- the thumbnail
- the filename
- the # layers, and consequently the start / end layer sliders don't works as expected
- .. probably more

sophiedeziel, 2024-01-11:
I've noticed that and gave it a try to fix this!

It seems like the library need to provide some callback hooks. Is there another approach you are thinking about?

I also started implementing a select dropdown in the demo to rapidly switch between the existing `.gocde` files in the folder. That's really useful IMO to play with different kinds of files. It works pretty well. I may open a PR if you like the idea.

remcoder, 2024-01-12:
hmm, maybe drag n drop shouldn't have been part of the lib actually. Or at least not as part of the WebGLPreview class. I don't think I would add it it there at this point. I'm definitely reluctant to expand the drag n drop code inside the WebGLPreview class. I guess a second event handler can be registered on the canvas, so this code can live in the demo?

Besides, I was actually thinking about re-writing the demo app. The current demo (which I host at https://gcode-preview.web.app/) has these design principles:
- demo the preview with all possible options
- plain javascript, no framework
- plain script tag loading (so not ESM)

But this feels dated now. Especially the last one because THREE.js no longer provides non ESM builds since recently. So this will have to be fixed anyway. This is actually a good thing, because I never dared to abandon non ESM builds for fear of breaking backwards compatibility. But since THREE has made the jump we only need to follow. 

And since the current demo code has a lot of boilerplate code it would probably be cut in half by introducing a UI framework. So this is why I was sorta planning on rewriting the demo in Svelte.

Another idea I'm having is to split up the app at https://gcode-preview.web.app/ from the lib. To avoid the double role that the demo now has. 

ok... sorry for the rambling ;-) 

I think what is best for the demo now is to try and fix it from within the demo code. If I later decide to rewrite it in svelte I'll just create a new folder or repo. This way no effort is lost.

sophiedeziel, 2024-01-12:
> maybe drag n drop shouldn't have been part of the lib actually.

My first impression is that this is right. It's the kind of feature that should be handled by the application using the library. The application may need access to the whole GCode, metadata, store the file, etc. It's not the previewer's responsibility to provide all of those.

That being said, are you thinking about a deprecation path or just leave the feature untouched? Deprecation makes sense because supporting it long term may become a burden. 

> Besides, I was actually thinking about re-writing the demo app.

> Another idea I'm having is to split up the app at https://gcode-preview.web.app/ from the lib. To avoid the double role that the demo now has.

Let me suggest to bring this a step further by also splitting the framework examples. 

I kinda like having the demo in the same repo, because it's simple to use that way. Development of features and usage by the demo can always be in sync and enforced via PR reviews. Wild idea: since we have access to types, could the demo dynamically generate the right input fields when options are added? Order, grouping, names, limits, validations, etc, could be defined explicitly and fallback to some default inputs. That would ensure that `demo the preview with all possible options` is always true even when the library and the demo are in different repositories.

I don't have an opinion on the framework to use or keep plain javascript. I think that the design principles should serve a specific goal. Like `demo the preview with all possible options` helps someone to make decisions on what library to use and figure out if it serves their need. Plain javascript may help as an example of usage implementation. But that is the role of documentation. A framework may help in developing quickly a great GUI to serve `demo the preview with all possible options`. 

Let me suggest some principle ideas for debate and ways to accomplish them

## Reliable
- **Up to date with the latest released version**
- **Demo the preview with all possible options**
- **New options inputs dynamically added**

## Useful
- **Provide ready to use diverse example files and config presets as well as support the user's files** (like a CNC specific profile and GCode, a vase with specific colors, a single layer with travel moves, different sizes and build volumes)
- **Built in documentation of each options** (could simply be links to the official docs) (mention of experimental modes, deprecations)
- **Export/import configuration option values** (To reproduce and document bugs or to simply design and use a color palette)

## Easy and fast to maintain
- **Deployment automations**
- **Some level of CI tools**
- **Easy to use locally alongside a local development version of the library**
- **Lightweight framework, architecture and dependencies**

## Delightful
- **Optimized for trying it out quickly** (snappy, smaller gcode, presets, limited loading time)
- **Balance between affordances, information and simplicity**
- **Information is easy to access** (3d files credit, docs, links to github/npm/licence)

# Getting back to the initial topic of this issue

Yep. A quick fix is possible without rewriting the whole thing. Old and new principles can be followed as progress is made.

remcoder, 2024-01-12:
Yes, deprecating the drag 'n drop feature is a good idea. I don't know of a standard way other than just logging a warning btw. There is currently no API reference either, something that needs to exist at some point. 

Maybe the cleanest way forward for fixing this is issue is to stop using using the deprecated DnD from the lib and moving it to the app side. This way the demo won't need to be updated if we ever remove the DnD code from the lib. Fixing the bug in de demo is then trivial.

I'll need to get back on your other points at a later time if you don't mind ;-)

sophiedeziel, 2024-01-15:
I am not aware of a widely adopted standard for javascript packages, but I know a good number of best practices for deprecations. As part of my paid job, I was responsible for planning for some breaking changes on a heavily used public HTTP API (at Shopify). How much a thing is used and how flexible the api clients are have a big role in the grace periods we decide to give.

I see here that the drag'n drop feature is off by default. 

https://github.com/remcoder/gcode-preview/blob/3d1babc887f112a3f70d3012150daceb202f92c9/src/webgl-preview.ts#L84

This is good, because the blast radius of this breaking change is reduced and warnings in logs can be targeted!

It would be a good idea to bundle together any breaking changes you want to make for version 3.0 . I think planning for a 3.0 release in 6mo or more is reasonable, but it could be shorter depending on the nature of all of those changes. They can be anything, like dropping an old node version support, upgrade three.js, drop non-module builds, etc. 

My suggestions
1. Define the version calendar 
     - When do you want to ship 3.0
     - What other breaking changes to bundle together, set a milestone
     - for how long do you want to also support security and bug fixes on 2.x when 3.x is out. Not at all is valid
2. log a warning when the feature is enabled, with explanation. Ship with next 2.x version
3. I've seen features extracted as a plugin in a separate repo. Worth considering for dev's happiness when they upgrade. But if it's not used a lot, or if you think that it's not worth it, you can skip.
4. Upgrade path in the 3.0 changelog (plugin installation, for example)

remcoder, 2024-01-15:
Agree on deprecating DnD for the next 2.x and remove in 3.0. If you feel it it'd be nice to have a 3.0 branch at this point in time let me know. And I think we can use a label to group tickets that target 3.0.

commit bfa3aeed (2024-01-16): Add deprecation notice for DnD as a lib feature. Also fixes #83
closed 2024-01-16 by remcoder