xyz-tools/gcode-preview #85: Possibility provide custom range for brightness

ivan-hryshko, 2023-11-30:
I have a problem with displaying black (#000000), gray(#808080) and white(#FFFFFF) they all look the same , 
Is it possible to set range for:
black color for example brightness in the range [0-30]%
for gray [30-70]%
And for white [70-100]%
I think it will make the display of models more realistic
[image]

remcoder, 2023-11-30:
Good point. It is a limition of the current implementation.

I'm trying to think of a way that works also for non black/gray/white colors. And I'm reluctant to add more config options but I'm also reluctant to change existing behavior.

Do you have some ideas?

Also to get this moving a PR and some test cases would be really helpful 😄

ivan-hryshko, 2023-12-03:
@remcoder  i dont have permision to make PR, but i. thing about something like this
 ```
 minBrightness: number = 10;
 maxBrightness: number = 70;`
```

```
    this.maxBrightness = this.validateBrightness(opts.maxBrightness) ?? this.maxBrightness;
    this.minBrightness = this.validateBrightness(opts.maxBrightness) ?? this.minBrightness;
```
 ```
validateBrightness(value: number) {
    if (value === undefined || value === null || typeof value !== 'number') {
      return undefined
    }
    if (value > 100) {
      return 100
    }
    if (value < 0) {
      return 0
    }
    return value
  }
```

 ```
     
        const preparedMaxBrightness = this.maxBrightness / 100
        const preparedMinBrightness = this.minBrightness / 100
        const brightness = preparedMinBrightness + (preparedMaxBrightness * index) / this.layers.length;
```

ivan-hryshko, 2024-01-10:
[https://github.com/remcoder/gcode-preview/pull/92](https://github.com/remcoder/gcode-preview/pull/92 ) 
@remcoder mode extrusion as tube look amazing it fully resolve my problem displaying black and gray colors.
When do you plan to add this mod to master? Can this mode be used and does it still need to be tested?

remcoder, 2024-01-11:
Tube rendering is under development by @sophiedeziel so let's give her some time to complete it. There is no fixed scope or schedule but you can follow the discussion in issue #89 .

You're free to use the current implementation of course but it will probably change still. For the time being I think you can reference the github repo and branch in your package.json. Please let us know of any findings when you encounter them!

remcoder, 2024-01-11:
update: tube rendering was [released as v2.12](https://github.com/remcoder/gcode-preview/releases/tag/v2.12.0) albeit still experimental. 

Can I close this ticket?

ivan-hryshko, 2024-01-12:
@remcoder i think yes

closed 2024-01-12 by ivan-hryshko