xyz-tools/gcode-preview #89: Render extrusions as tubes

sophiedeziel, 2023-12-18:
I've been playing a lot with the code and ended up with a prototype that renders extrusions as tubes instead of lines. The idea is to be able to render a more advanced preview that resembles what PrusaSlicer and Cura have. I was wondering what are your plans for future versions of this tool because this is not a minor change and is worth planning carefully. 

Here is a screenshot of what I was able to render:

[image]

Extrusion paths are tubes and travel moves are still lines.

I was thinking about switching from lines to tubes using a boolean in the preview options. 

However, my prototype is very far from performant and is not accurate enough in my opinion. I'm still learning Three.js and feedback is welcome.

If I submit pull-requests, how would you prefer to proceed? Of course, smaller PRs with a clear scope are preferable. Is there a list of options and calculations that matters to you? Here is what I had in mind:

- Option to switch from lines to tubes
- When the metadata is available, different colors for outer lines, inner lines, and infill
- Accuracy in the dimensions of the tubes for layer height and line width
- Transparency of travel move lines

Let me know what you think. There are a lot of cool things I can add to this project and I'd prefer to contribute here rather than using a fork.

remcoder, 2023-12-19:
Wow the screenshot looks really good! 

To answer your question first: I'm not dedicating any time to developing on the project myself atm. This is mostly due to time and life priorities. That being said, I've always answered issue's and PR's. And sometimes this sparks a moment where add some improvements myself. 

So I guess if you want to use this code as a base, please go ahead. Maybe if you want to go really fast, a fork of your own might be the best way to go. If you contribute here, on the other hand, I, and others who follow this repo, can assist you here and there. 

I would personally really love if you collaborate on this project, and I'm open to give you access to the repo. Just remember that I cannot make any promises about my own time.

About the tubes: I have several things I'm thinking about:

Rendering extrusion paths with volume can be done in a few ways, including round tubes. Maybe the simplest to start with would be to use square tubes. This results in a minimum of triangles in the mesh, as opposed to round tubes. And some basic things like line width and centering would need to be tackled anyway. 

Another route would be to use the [model used by Slic3R / PrusaSlicer](https://manual.slic3r.org/advanced/flow-math). 
[image]. It's a little more complex and uses more triangles.

Actually, AFAIK Cura uses square tubes in their flow calculations, resulting in less spacing between the lines than in PrusaSlicer, for the same line width. (in fact [OrcaSlicer now has a param to for a third option that sits in between](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.8.1))

What I'm getting at is that the best way to render the extrusion lines actually depends on the slicer that was used.  ... but I'm probably getting ahead of myself here ;-)

I hope to have answered your questions mostly. If not, shoot 😃

sophiedeziel, 2023-12-19:
I'll start off by saying that I totally understand the time limitation and the fact that we are all humans with a (hopefully fulfilling) life. If we have compatible and vision, I may be interested in contributing long term to this project and help you with the tasks related to maintaining an open-source repository. But first, let's build some trust and see how we can make this first step work out!

I was able to clean up my code yesterday, and I have a first draft of the feature. However, it is not production ready in my opinion. I optimized slightly the code, but there's still a lot to do. I was able to get a render time of ~6 seconds and fluid camera movements on a 2019 Macbook Pro. It is reasonable if the user is provided some feedback on the progress, but it's something that need to be implemented.

If you are curious, here is my branch:
https://github.com/remcoder/gcode-preview/compare/develop...sophiedeziel:gcode-preview:tube-extrusion

It is a good candidate right now for a first PR with fast follows for improvements. I had to modify the signature of some methods and classes and there's some things that are not good enough to my liking. But it is a good first step. 

What you have in mind with the Prusa slicer model is my ideal strategy. I'm not sure yet how I might implement it in a way that remains performant. I may use a tube that has a diameter equal to the extrusion width and truncate it, but I want to focus on optimizing and provide enough user feedback right now.

sophiedeziel, 2023-12-20:
Yesterday, I focussed on performance and here are some of my findings!

My first prototype used a single square tube for each vector. The `catmulrom` curve improves rendering time by a lot. It seems like it's the number of geometries that matters in our case, not the complexity of each one of them, which is why I decided to combine them in this way. The result looks way better with the rounded corners, but it's an actual performance improvement.

I also experimented with some different materials. the `MeshLambertMaterial` is the fastest, but the difference is not that significant.

I tried updating to three.js to the latest version and the materials don't look the same at all, shadows disappear and the layers are impossible to see distinctively. But I did not notice a significant performance improvement.

The render method spends most of its time in `new TubeGeometry(curve, curvePoints.length * 50, 0.4, 6, false)`.  The more I add to `curvePoints.length * 50`, the longer it takes, but it's still marginal. Going as low as 1 or 2 does not remove seconds. 50 is the threshold that I've identified with that model for rendering flat layers correctly. Developing an algorithm for it is a longtail optimization.

One approach I would like to take is to generate the geometries in the background and cache them. That approach would also work for rendering the lines. Switching modes would be fast. To validate this, I tried storing the geometries groups in separate variables and switch them back and forth without regenerating the geometries. It was fast, under 500ms. However, that approach would only be a good one if the realistic extrusion mode is disabled by default or if there is a progress bar showed to the user. 

I'm about to create some benchmark tooling! I think it is worth it for the next steps.

sophiedeziel, 2023-12-20:
I double checked and I made a mistake in one of my claims:

> The more I add to curvePoints.length * 50, the longer it takes, but it's still marginal. Going as low as 1 or 2 does not remove seconds. 

The performance is significantly better as the number of tubular segments goes down. But at the cost of accuracy. 

I took some screenshots as a demonstration. The first one is all layers, and the dev console output of the total render time.
The second one is a close up of layer 4.

## *1 multiplier

Render time is 229ms. As a reference, the line render is around 45ms
[image]
[image]

## *5 multiplier

We can see the overall shape is significantly better, but single layers are really bad.
[image]
[image]

## *25 multiplier

We're almost there for single-layer mode
[image]

[image]

## *50 multiplier

[image]
[image]

## Some optimization ideas

I'm considering multiple approaches based on those learnings.

Single-layer mode is significantly easier for the renderer. The quality and precision of that mode could be higher than when the whole model is rendered. Based on that, there's a lot of shortcuts the full model mode could take:

- When it's possible to know what is outer lines, inner lines and infill, the whole model view could render outer lines first for fast preview, and add inner and infill in separate steps.
- Inner and infill could also have different values for the number of tubular segments. 
- We could take an approach that gradually increases the number of tubular segments, starting with a low value for a fast first render and increasing it as the animation progresses
- Dynamically adapt the number of segments based on the length and number of angles of the lines
- Provide some user control on the variable for render speed/quality

I think that `Dynamically adapt the number of segments based on the length and number of angles of the lines` is the best approach for a second iteration of this feature.

sophiedeziel, 2023-12-20:
Sorry for the number of posts. But I'm making a ton of progress in a short period of time.

I simply replaced 

```Typescript
const geometry = new TubeGeometry(curve, curvePoints.length * 50, 0.4, 6, false)
```

by

```Typescript
const segments = Math.floor(curve.getLength() + 2) * 2;
const geometry = new TubeGeometry(curve, segments, 0.4, 4, false);
```

and I got amazing results!

[image]
[image]

Render time is down to <500ms, which is very acceptable, and complex layers are even better!!

I'm going to open a PR now.

remcoder, 2023-12-20:
Just a heads-up, I just merged some work I was busy with which will cause merge conflicts but nothing major. I chopped up the render function and added a first unit test relating to the gcode processing (both long overdue). Also, the unit tests now run as GH action so overall this is for better quality/stability going forward.

sophiedeziel, 2023-12-21:
That's awesome! 

I already resolved the conflicts and it seems like the test is still passing.

I'm still unfamiliar with typescript's ecosystem and testing frameworks. I should be testing my work too, I'll dive into it.

remcoder, 2024-01-11:
Hey @sophiedeziel, pinging you for the status of this issue and when to release. Might as well do a release soon/now maybe? What are your thoughts/plans on this?

sophiedeziel, 2024-01-11:
I am experimenting a lot to improve performance and I noted a lot of potential improvements. I think performance could be it's own specific GH issue and follow up PRs if we want to release what we have right now.

As it is right now, there are no bugs to my knowledge related to the tube rendering features. No software is bug free, we just don't know about them yet. But nothing is an obvious blocker for releasing.

Do you want to scope in the dynamic tube's radius as part of the next release? I don't think it's needed and I don't think we should rush it. Also should be it's own GH issue to talk about things like API and calculation based on layer height or line width.

Would you prefer an explicit mention that the mode is experimental or you feel it's good to go? Advantages of keeping it in an experimental mode is that we have a bit more freedom to change APIs and behaviour, while still making it available for people to use it.

remcoder, 2024-01-11:
thx for chiming in @sophiedeziel. 

I created separate tickets for rendering performance (#95) and dynamic tube radius (#96)

Fully agree there are no blockers for releasing as experimental but I wanted to check with you. So I'll now go ahead and create one.

remcoder, 2024-01-11:
This is now [released as v2.12](https://github.com/remcoder/gcode-preview/releases/tag/v2.12.0)

closed 2024-01-11 by remcoder