xyz-tools/gcode-preview #98: Create benchmark tools

sophiedeziel, 2024-01-12:
I'm looking into creating something to automate benchmarking performance with different cases. 

There are several approaches I can take and I'd like to discuss them here.

## Options

### Option 1: In browser

What I have in mind is something similar to the demo that can be used in browser. 

Pros:
- Easy to use and visualize
- Access to the browser's dev tools to help debug
- Can interact with the library directly to "feel" the changes
- May be easier to implement visualizations (like flamegraphs and tables)

Cons:
- Heavier to code and support
- There may be differences between browsers

### Option 2: CLI tool

I'm hoping that the testing framework could help into watching files and run benchmarks

Pros:
- Fast to implement
- Can be used as automated CI requirements (issue warnings of fail on significant performance drop)

Cons:
- Boring
- May be challenging to set up initially given we're interacting with canvas, 3D environments and WebGL 

##

My opinion is very biased by the fact that I'm working on this project mostly because I think it's fun. So I lean towards Option 1 because I enjoy the process of making something delightful to use.

remcoder, 2024-01-12:
Just to note, I think it is the performance in a browser context that really matters. Anything else would be artificial and might lead us to go chasing ghosts :-)

btw, benchmark tooling would only serve to help the work on performance, right? So I think your preference is most important. Just go with what feels best to you.

Anyhow, I started this project for fun, so go have some fun! :-)

sophiedeziel, 2024-01-12:
Yes, it would greatly help to work on performance, as well as help identify any regressions with new PRs. 

It has the potential to evolve in development/debug tooling for new features but I don't want to fall into the scope creep trap before I'm even started!

remcoder, 2024-01-12:
ok, so what is your focus? What do you want to measure?

sophiedeziel, 2024-01-15:
The very first iteration I want to implement is render time comparisons for a small number of GCode files. The comparisons could be tube vs lines, and local dev version vs last published version. Absolute time measures vary with machine and browser, so it has to be compared to something.

Those benchmarks should target very specific conditions, so I would create gcode files specifically for those.

- A Benchy for a general idea of performance variations
- A shape that results in lots of short travels and small extrusions (maybe 10 small cubes)
- a cube of 100% infill to test layers that have a lot of points
- A vase (that would simply be a cylinder), to test a very long segments with varying heights
- Something to test re-renders (ie change colors or switch between render modes)

I'm taking this approach because I have some ideas on how to improve rendering times in most of the scenarios above. 

Times could be split, with details of time spent parsing the GCode, splitting in layer points, building the geometries and adding them to the scene.

sophiedeziel, 2024-01-15:
I wonder if the benchmarks should be a separate repo. You mentioned that Svelte could be used for the demo app, and I think that sticking to the same stack is easier to manage. So the benchmarks should also be built using Svelte (or whatever is picked). If it's using a framework, separate repo is a good idea. Or we can split later?

remcoder, 2024-01-15:
I would be so happy with something as simple as an html page with 5 or 6 different gcode files and render times. That alone would already be super useful. 

That and some rendering options, like lines vs tubes, that can be toggled for all instances at once.

Your first iteration could be as simple as that and already be super valuable 😄

remcoder, 2024-01-15:
At this point I'd like to keep stuff in one repo as much as possible. Maybe the benchmark can be done in vanilla js? I would stay away from frameworks that require build systems for this purpose actually. Svelte is impossible to use w/o a build system actually. Vue.js could be an option and I don't know if React can be use via only a script tag..

remcoder, 2024-01-15:
btw, how about we drop some of the discussion in Discord? I've setup a brand new server:  https://discord.gg/3wjx5SVc

remcoder, 2024-02-01:
@sophiedeziel you mentioned elsewhere that you might not need the benchmark tooling at this point to support the performance work. Shall we close this then?

sophiedeziel, 2024-02-02:
Yes! Let's close!

closed 2024-02-02 by sophiedeziel