xyz-tools/gcode-preview #100: Some travel lines renders as extrusion lines

o0Ghost0o, 2024-01-16:
Hi, I've encounter a bug when setting the lineWidth to anything, if I leave the GCodePreview.init({}) settings without the lineWidth it won't render those travel lines

Without the lineWidth parameter
[image]

With the lineWidth parameter to 1
[image]

remcoder, 2024-01-16:
thx for reporting @o0Ghost0o. I'll take a look at this.

closed 2024-01-16 by remcoder

remcoder, 2024-01-16:
@o0Ghost0o this is [fixed in v2.13](https://github.com/remcoder/gcode-preview/releases/tag/v2.13.0).

o0Ghost0o, 2024-01-17:
@remcoder, nice, wish all bugs were like that xD. Thank you!, btw, I can give you an example for vue3 using <script setup> too (need to make the template as I'm using it with laravel).

remcoder, 2024-01-17:
@o0Ghost0o that would be awesome! 

Not sure what you had in mind but I think you could maybe either send a PR to:
- add a new vue example
OR
- add a section to the README

Feel free to [open a ticket](https://github.com/remcoder/gcode-preview/issues/new/choose) too if you want.

PR #103 (2024-01-16): Fix incorrect segmenting of extrusion lines when line width is set.
Fixes #100
commit de1f9872 (2024-01-16): Fix incorrect segmenting of extrusion lines when line width is set. Fixes #100