xyz-tools/gcode-preview #120: extrusionColor: Ability to set colour at extrusion line-level

tom-walkr, 2024-02-21:
It would be incredibly useful to be able to set extrusionColor selectively for each extrusion line. The previewer could take an array of hex strings as an argument and map these to corresponding extrusion lines.

This would enable selective colouring of G-code features (walls, infill, etc.), and give greater aesthetic control. 

Myself and my colleagues can work on this, but if anyone is working on this already, or has a plan to, please shout!

sophiedeziel, 2024-02-21:
I wish that would also be possible and have some ideas on how to do it.

First of all, the internal structure and API assume layers and have them built in, but rigid attributes for color. I don't think this should stay long term, but it's the most common paradigm today and making breaking changes on this is a very hard sell. I don't think it's the responsibility of the library to be able to parse slicer-specific metadata for distinguishing infil, walls, etc, but it already has a precedent.

Without breaking everything, I have ideas on how to manage chunks of gcode by passing arguments instead of letting the library calculate everything. I mentioned it on the project's Discord last week 😄 .

I'm playing with the idea of a re-architechture and plan on building a prototype soon, which would enable this specific feature request. I'm really curious about how you envision this. Would the mapping be `zip` style or you're thinking about passing line numbers along with the hex? Maybe labels? 

I think that this feature should not be rushed and many designs should be considered.

tom-walkr, 2024-02-21:
Agree the library should not be responsible for parsing slicer-specific data - responsibility should be on user here, as too complex to cater for all configurations.

My initial thought was passing in an array with a `zip`-like mapping, but now that you mention it, I see merit in the approach whereby line numbers are passed in with a hex. From a user's perspective, generating these inputs would likely be quicker and easier for small sections. All lines not mentioned in array would then take on the 'global' `extrusionColor`.

remcoder, 2024-02-21:
Maybe I'm missing something but I'm not yet sold on passing, from what I understand, a huge data structure to the lib. Wouldn't this require the consumer of the lib to do their own parsing of the gcode?

I'm all for protecting the responsibilities of the lib but I also see merit in adding some logic to the parser to interpret comments like `;TYPE:Skirt/Brim` to color them accordingly. 

So whenever the line `;TYPE:Skirt/Brim` is encountered the current extrusion color would switch to the color for brims. And so on. 

The parser can translate such lines into a new command type for this purpose, say a SetColorCommand. This would even work nicely when we come to rendering multi-color prints, solving it in the parser.

To make it more slicer-agnostic a dictionary could be passed that maps strings (or regex's) to colors. I'm also not against collecting several  such dictionaries for different slicers in a contrib folder.

sophiedeziel, 2024-02-22:
> To make it more slicer-agnostic a dictionary could be passed that maps strings (or regex's) to colors.
>
> ```javascript
> {
>  ';TYPE:Skirt/Brim' : '#ff0000',
>  ';TYPE:Perimeter' : '#00ff00',
>  ';TYPE:External perimeter' : 'pink'
> }
> ```
> I'm also not against collecting several such dictionaries for different slicers in a contrib folder.

I love that approach. It's the perfect middle ground IMO. With that as a starting point, I think it's important to consider when blocks have starting and ending lines and if we want to go wild and allow nestest blocks. I may be carried away with enthusiasm, but imagine:

```javascript
[
  {
    'blockStart': \;MESH:box.stl\,
    'blockEnd': \;MESH:NONMESH\,
    'rules': [
      {
          'blockStart': ';TYPE:WALL-OUTER',
          'blockEnd': \;TYPE:*+\,
          'color': 'pink'
      }
    ]
  },
  {
    'blockStart': \;MESH:lid.stl\,
    'blockEnd': \;MESH:NONMESH\,
    'rules': [
      {
          'blockStart': ';TYPE:WALL-OUTER',
          'blockEnd': \;TYPE:*+\,
          'color': 'orange'
      }
    ]
  },
  {
      'blockStart': ';TYPE:WALL-OUTER',
      'blockEnd': \;TYPE:*+\,
      'color': 'blue'
  }
]
```

I know it may be overkill.

But it allows for having a pink outerwall for the box, an orange lid, and everything else on the build plate is blue.

remcoder, 2024-02-22:
Well we could start with blockStart and color and leave the door open to a possible extension.
```
{
      'blockStart': ';TYPE:WALL-OUTER',
      'color': 'blue'
}
```

We'd only need blockEnd if we're nesting right? 

I'm skeptical about a usecase where you both want to color objects and line types. Let's also think about what we want to visualize. If we want to identify walls, it makes sense to have them all of the same color so as to easily spot them all. So from a UX perspective, I'm leaning more towards coloring either lines types or objects (or extruder number) but not a combination. 

@tom-walkr was there a particular reason you suggested setting colors at the line-level? From the mentioned solutions would work for you?

tom-walkr, 2024-02-22:
Our use case is to colour lines so we can geometrically display print quality information that we generate during printing, which I admit is rather specific!

We have actually achieved what we need the library to do by adapting `addThickLine()` to take an array of colours (mapped to cartesian coordinates), generated by us, to selectively colour each vertex.

However, the ability to selectively colour different line types would also be very useful, and I would support the idea to add this as functionality to the public repo.

sophiedeziel, 2024-02-22:
I'm bringing back the point of rendering colors based on something that is not found directly in the GCode, but passed in as line numbers, or something else.

I have some use cases that will help understand.

- Let's say I have a text editor and the preview side by side. It would be nice to have lines that are selected in the editor highlighted in the preview. `{ 'startLine': 102, 'endLine': 124, 'color': 'blue' }`

- Another use case would be that the user has a collision detector of printed parts with the hotend. Those segments could be highlighted in red by passing a similar array of segments to color.

- Or bridges longer than a threshold. 

The idea would be to let the user define the logic for colors, and the lib to define the interface for the set of rules, priority and a polyvalent structure for start/finish

sophiedeziel, 2024-02-22:
On a different aspect of this feature, do we want to also consider applying colors after the gcode is rendered? There could be some public methods to reset colors and apply them. 

```javascript
const preview = new GCodePreview.init({...});
preview.processGCode(some_gcode);

function handleLinesSelected(lineStart, lineEnd) {
  preview.clearColors();
  preview.addColorRule({
    lineStart: lineStart,
    lineEnd: lineEnd,
    color: 'light blue'
  });
  preview.render();
}
```

remcoder, 2024-02-22:
Trying to summarize then:
- no need right now to color every line individually
- would be nice to color _some_ lines by _line number_, as per above.
- coloring per _line type_ is definitely wanted but should be a separate issue.

please correct me if I'm wong

remcoder, 2024-03-15:
Closing this issue for now as there seems no pressing reason to start development on this. 

Feel free to re-open if you think otherwise.

closed 2024-03-15 by remcoder