xyz-tools/gcode-preview #121: set extrusionColor for different extrusion types (wall/infill/etc)

remcoder, 2024-02-22:
Slicers produce different 'types' of extrusion, or rather, these are generated by different subprocesses of the slicing process.

For instance, perimeters are generated first after which the infill is generated. It is possibly to have the slicer output metadata (formatted as GCODE comments) to mark the different types of extrusion. 

This enhancement discusses the options for rendering the different types with different colors.

As a strawman consider  the following pseudo-code:

```
const colorInfo = {
      'blockStart': ';TYPE:WALL-OUTER',
      'blockEnd': \;TYPE:*+\,
      'color': 'blue'
  }
const preview = GCodePreview.init({
      canvas: document.querySelector('canvas'),
      extrusionColor: colorInfo
  });
  
```

sophiedeziel, 2024-02-22:
Could `colorInfo` also be an array for different types?

remcoder, 2024-02-22:
But it is ;-)

remcoder, 2024-08-15:
I think this is outdated now that we already have multi-color previews for gcodes with tool switching. I see color coding as a single dimension axis so I propose to introduce mode switching to change from multi-tool to line type visualization. 

## Approach 1: overloading extrusion color
One approach could be just to add a specific mode param which causes the extrusionColor param to be interpreted differently. For example:

For visualizing multi-color / multi-material:

```ts
mode: 'multiTool', // default, so optional
extrusionColor: ['red', 'green', 'blue'] // color T0 as red, T1 as green , T2 as blue
```

And for visualizing lines types:
```ts
mode: 'lineType',
extrusionColor: {
 'perimeter': 'red',
 'infill': 'green',
 'overhang': 'blue' 
}
```

This is slightly awkward because the type of the extrusionColor param changes depending on the value for the mode param. Although possible, it will lead to awkward code here and there.

remcoder, 2024-08-15:
## Api Approach 2

Extending the extrusionColor value to more descriptive object leads to:

### 2a

```ts
extrusionColor: {
  mode: 'multiTool',
  colors: {
   t0: 'red',
   t1: 'green',
   t3: 'blue' 
}
```

```ts
extrusionColor: {
  mode: 'lineType',
  colors: {
   perimeter: 'red',
   infill: 'green',
   overhang: 'blue' 
}
```

or

```ts
extrusionColor: new MultiToolPalette ({
 t0: 'red',
 t1: 'green',
 t3: 'blue' 
})
```

```ts
extrusionColor: new LineTypePalette ({
 perimeter: 'red',
 infill: 'green',
 overhang: 'blue' 
})
```

The underlyging
The latter has less strings and should be little less error-prone in a non-ts conext