xyz-tools/gcode-preview #123: Collect flavours of multi-color (/ multi-material) gcode

remcoder, 2024-03-08:
We want to support multi-color prints (#120). To get started we need an overview of the different flavours of multi-color printing and a few example gcode files to go with that.

- T1,T2,...Tx (tool change) 
- M600 (filament change) 
- ..

remcoder, 2024-03-08:
Thinking out loud here:

- Tx: requires mapping of color per _index_ (the x) ex `{ 1: 'red', 2: 'green' }`
- M600: requires a sequence of colors. The sequence may need to contain repetition: ['red', 'green', 'red']
-

sophiedeziel, 2024-03-09:
I'll generate some as soon as possible.

remcoder, 2024-07-27:
I have the feeling this isn't so much of an issue really. Ok if we close it?

sophiedeziel, 2024-08-03:
Let's do it. There are many other ways to learn more.

closed 2024-08-03 by remcoder