xyz-tools/gcode-preview #142: Purge Line does not seem to be rendered

kyleisah, 2024-04-28:
Hey there! 

I'm working on a simple little python script that will help users generate a nice beefy line purge to prime the nozzle prior to printing.

I can see the preview of the line purge in my typical slicer, but when I try to render it in your gcode previewer live demo, it doesn't seem to show up. Not sure if it has something to do with where gcode starts being read, but figured it should be mentioned.

Here's the gcode rendered in Prusaslicer's previewer:

[image]

And here's how it's rendered in your live demo:

[image]

I'll also drop the sliced gcode file here, as a .txt,

[Shape-Cylinder-9.52316g-24m.txt](https://github.com/remcoder/gcode-preview/files/15141077/Shape-Cylinder-9.52316g-24m.txt)

Thanks so much! If you need anything else, let me know!

remcoder, 2024-04-28:
Hi @kyleisah, thanks for bringing this to our attention. We'll be taking a look at it soon.

remcoder, 2024-04-28:
Just found out that the purge line isn't rendered bc GCode Preview doesn't like spaces right before the numerical params.

So, this original snippet doesn't render a line:
```
G0 X95 Y 99.2894
G0 Z 0.8
G1 X155 E 50.0  F 718.42
```

While this version with the spaces removed _does_ render a line:
```
G0 X95 Y99.2894
G0 Z0.8
G1 X155 E50.0  F718.42
```

Should be fixed soon in the parser. As a temporary solution you can remove the spaces for now.

kyleisah, 2024-04-28:
Nice work! The additional spaces are something the script seems to be doing, I'll see if I can add some stuff to remove trailing spaces after the axis and feedrate denominations are printed out. Thanks again!

kyleisah, 2024-04-29:
Sweet, fixed my script. Was escaping the print function to print variables and it all got so confusion. fstrings to the rescue!
[image]

remcoder, 2024-04-29:
That's great! And thank you for reporting back. 

I'll keep this open so we'll fix this issue in the parser at some point.

Good luck with those beefy purge lines! 😉

closed 2024-05-01 by remcoder

kyleisah, 2024-05-01:
Cheers!

PR #143 (2024-05-01): Allow spaces in gcode params
this fixes #142