xyz-tools/gcode-preview #152: Support M600 filament swaps

remcoder, 2024-05-29:
Now that we have multi-color rendering to propery preview multi-color prints the API for extrusionColor was updated to allow an array of colors.

ex. `extrusionColors: ['#CF439D', 'rgb(84,74,187)', 'white', 'rgb(83,209,104)'], `

But there is another way to use muliple colors for a print, which is a filament swap. This works by manually or automatically changing the filament at a certain layer. In GCode this is controlled via the [M600](https://marlinfw.org/docs/gcode/M600.html) command.

This technique enables for instance multi-color sings and even color mixing in 2d (hueforge)
[image]
[image]

The obvious API for this sequence of colors would also be an array. But the question then is: how to define a possible combination of sequential and concurrent colors? Nesting arrays would be confusing.

For tool based multi-color it is known in advance which color is associated with which too. At least that correspondence doesn't change during the print.

For filament swaps, you cannot know how many swaps there will be. The whole gcode stream will have to be processed. And it just depens on what tool is active at that moment. That makes filament swap much more dynamic and less predictable. 

Then there are systems, like the AMS from Bambulab that will allow you to continue on the next roll when the first is used up. Since GCode preview doesn't know anything about spools of filament it is impossible te predict when and where to do a color change without extra information. Therefore, I would suggest that this type of filament swap is out-of-scope for this ticket.

So supporting M600 but not sensor based filament swaps, is the focus.

My first suggestion for the api is to define an extra array: `swapColors: []`. Whenever an M600 is encountered, the next color in this array is used.

sophiedeziel, 2024-05-30:
I think we are too focused on rendering the actual color of the filament. 

There's an expectation of accuracy when visualizing gcode but sometimes we don't have the information and we should abstract it out. 

Let's think about representing the changes with color indexes rather than trying to map colors to the actual world! My slicer, for example, does not care about what colors I loaded in to each extruders. Tool 1 is always orange, tool 2 is pink, etc. 

Before the gcode is loaded, we can't know how many color changes there will be. But we can loop through a default array that indicates that the color changed. In the case of M600, an array of two is enough to understand what is going on during the print. If the user wants to make it pretty, they can add more colors to the array. It's pretty unlikely that the array will be longer than 10 colors, different or duplicated, because it's a manual change that would be super annoying in the first place.

sophiedeziel, 2024-05-30:
Why would we use a different param than the existing `extrusionColor`? It is an array, and we can loop through it. We don't have to have it indexed and mapped perfectly to the number of M600s in the file.

sophiedeziel, 2024-05-30:
>but which color would we pick then, from the extrusionColor arrray?

I'm thinking about naively starting at index `0` and increment at each `M600` until it's time to loop back to `0`. So if there are two colors, it will just switch between the two. I'm not sure about what to do when there is only a single color. Should we have a default colors list to fallback to? Or slightly change the brightness? Do nothing?