xyz-tools/gcode-preview #156: Use clipping for the min/max layers iso re-rendering

remcoder, 2024-05-31:
see: https://threejs.org/examples/webgl_clipping.html

sophiedeziel, 2024-06-01:
It's safe to assume min/max layer parameters are controlled most of the time by a slider. The performance of it is very important for the user's experience. It can be difficult to diagnose a problem in the sliced model when the preview lags.

Can clipping be used for lines or is it a geometry thing only?

remcoder, 2024-10-18:
will be implemented in #234

sophiedeziel, 2024-10-18:
I was thinking that clipping would be also useful in other contexts than inspecting layers. For example, have the print split vertically. 

But that's a whole new feature that happens to use the same techniques. WDYT?

remcoder, 2024-10-18:
I could be useful but I haven't come across a situation where min/max layers weren't adequate. I'd like to see an actual use case. What would you use the feature for?

sophiedeziel, 2024-10-19:
You're right that we can visualize pretty much everything with the layers. But I've seen people struggle with 2D/3D vision and mental representation of objects. They could not understand from layers that there was overhangs.

Having an additional clipping plane is an additional tool that can help understand what's going on when your brain is not wired like a software engineer's. 

Those are also ways I would personally use it:
- inspecting how layers are placed on top of each other and detect weak points 
- inspect the insides of a hollow part to see where the walls have a thickness of only one line or some infill
- just playing with it because it's cool to visualize

remcoder, 2024-10-19:
Good points. I use 'section analysis' sometimes in Fusion which really is the same thing. 
I must say I'm from the 'less is more' school    but I'm open to a simple implementation of this.

Maybe only allow clipping planes along the x, y, z axes? 

For me this would still be a 3.1++ feature.. but hey they're more like guidelines not rules https://imgflip.com/gif/1suyyk

sophiedeziel, 2024-10-19:
Yep, section analysis is different than layers, and should be scope for "eventually".

Only on x,y or z is exactly what I had in mind.

closed 2024-10-22 by sophiedeziel