xyz-tools/gcode-preview #186: idea: preview multiple gcodes at once

remcoder, 2024-08-12:
This ticket aims to have multiple gcode files to be previewed in a single GCode Preview instance at the same time. 

## Why
This is useful systems that can process multiple gcodes in parallel.  Although niche, I think it's interesting and fun idea and it ties in with recent projects of the likes of ProperPrinting and Nathan Build Robots so this feature could be useful for those communities.

## How
Here's some points that need to be addressed:
- the gcode processing state needs to be reset (z = 0 etc)
- the scene needs to be kept

## Challenges
If the scene needs to be re-rendered somehow, all the gcodes need to be rendered again. (could be kept out of a first POC)

sophiedeziel, 2024-10-17:
Some thoughts:

The `Job` object introduced for 3.x is perfect to represent the different models. 

My advice would be to keep `preview.job` private because we'll probably need to instead use `preview.jobs`. This will allow to make that change without breaking lib clients.

remcoder, 2024-10-17:
What part of this is a must for V3? This touches upon the lifecycle issue I think. And if we don't flesh out multiple gcodes now, and make the architecture support it, we'll need another potential breaking change to eventually implement it.

But it is also a bigger, non-trivial feature to implement. I guess it comes down to our appetite for this in the short term

I'd be ok with keeping this for V4 if it does turn out to somewhat complex.

sophiedeziel, 2024-10-17:
Lifecycle has to be properly managed, that's for sure. And there should also be an abstraction level between the job and the scene that would hold multiple build plate objects and jobs. 

We can definitely fit that into v3 without breaking changes, we just have to be mindful about what is an internal concern and what is exposed. I'm suggesting to not expose the job for now so we're not breaking it then :D

It's non-trivial and requires a lot of the refactor we already planned. Because of those dependencies and effort, this would be a long term goal. Maybe by the time we get there, v4 will be around the corner. If we get to the point this feature is reasonably feasible, it means we did an amazing architecture job.