xyz-tools/gcode-preview #189: Wireframe render is incomplete

remcoder, 2024-08-16:
[image]

sophiedeziel, 2024-10-13:
I have a suspicion this bug has the same root cause as https://github.com/remcoder/gcode-preview/issues/216

It's all about lines rendered. 

I've been investigating for a while, trying to eliminate causes. `_wireframe` is just a boolean passed to the material and does not have logic attached to it. I've ruled out rerender logic, as it happens also on the initial render. 

I've tried playing with the WebGLRenderer's and scene settings, as well as changing material type and line settings. It happens anyways.

Could it be a bug in Three.js?

sophiedeziel, 2024-10-19:
There's an interesting issue in the Three.js repo: https://github.com/mrdoob/three.js/issues/29045

remcoder, 2024-12-27:
[image]

remcoder, 2024-12-27:
Since the issue in three still exists and this was merely a debugging feature I vote for removing it for now.