xyz-tools/gcode-preview #209: Request for home assistant lovelace card

Dronedale, 2024-10-05:
It would be awesome to port this project into a home assistant lovelace card, where you could control inputs using home assistant entities! 
Would this be possible?

remcoder, 2024-10-05:
Hi @Dronedale, that should be possible if the Bambu integration gives you the gcode at the current time in the print. The card would need to update the preview incrementally. There's a method `processGCode` that does just that. It accepts lines of gcode that will get added to the current preview. 

Don't know what framework Lovelace cards are written in but there are examples for react/vuejs/svelte to help you get started.

Dronedale, 2024-10-05:
Hi @remcoder,

Thanks for your quick reply, I would just load the whole gcode via sftp from the printer when the print starts, load the model into gcode-viewer and set the the endlayer provided by bambu integration.
Just saw that it might be possible to use vuejs with custom cards in home assistant. I will take a closer look at this.
Btw your project is awesome!

closed 2024-10-07 by remcoder

Dronedale, 2025-02-13:
Hi remcoder,

Just wanted to ket you know that I finally got this working using litelement library for a custom card in homeassistant.

At the moment this is just proof of concept, but if you don't mind I would like to publish a repo on github soon for this custom card reusing your code for managing the heavy stuff.

remcoder, 2025-02-14:
Hi @Dronedale,

Really cool you got this working! You can of course reuse our code however you want but it would be best if you can use gcode-preview as a lib instead of copying over some code.. If that doesn't work easily I agree just Making It Work is the first prio.

If you publish the repo I could have a look it too, which may be helpful.

Thanks for reaching out btw!

Dronedale, 2025-02-15:
Hi,

very nice, I will try to get everything working and then share my code with you for review. 
Actually i am using an older version from gcode-preview lib. The newest ones I could not get to work until now. 
I needed to slightly modify your lib, so unfortunately I could not implement it in package.json directly. Needed to access the renderer object within my module to apply the domelement to my html element of the card.
Will let you know when I'm ready