xyz-tools/gcode-preview #216: Some lines are not rendered properly

sophiedeziel, 2024-10-11:
The only way I found to reliably reproduce is using the CNC preset, but it also happens sometimes when rendering travel lines.

[image]

It is hard to see, but it seems like the lines are on the scene. They are just so small we can't see them. We can notice some green pixels.

remcoder, 2024-10-18:
This might be since the bump to threejs r169. Let's check that first.

sophiedeziel, 2024-10-18:
I tried downgrading the lib multiple times, I'm not sure it's that.

I think you should try because there is probably something I did wrong

remcoder, 2024-10-18:
Found the problem: setting the lineWith is broken when using r165+

PR #237 (2024-10-18): Fix thick lines and set min width to 1
fixes #216 

Apparently the lineWidth for 'fat lines' in threejs was changes in r165.
closed 2024-10-18 by remcoder