xyz-tools/gcode-preview #223: Option to render retractions

sophiedeziel, 2024-10-13:
I had that idea after investigating stringing issues that fused parts of a print-in-place flexible model. 

Prusa Slicer has the option of rendering retractions. It's useful to be able to render that. 

[image]

For this feature to be as accurate as possible, we'd have to implement the `G0`/`G1` with negative `E` case as well as introducing [`G10`](https://marlinfw.org/docs/gcode/G010.html) support.

remcoder, 2024-10-18:
Can I suggest we do this in v3.1+?

sophiedeziel, 2024-10-18:
Yes!

remcoder, 2025-01-14:
I just realized PR #269 actually detects and count retractions now.

atm, zero-length movements are removed from the scene because they would be invisible anyway. But they might as well be rendered and the interpreter at least detects them.

sophiedeziel, 2025-01-15:
There's also a native gcode retraction code, but I think slicers mostly use `g0/g1` for backward compatibility with older printer firmware.

remcoder, 2025-01-16:
Good point. I updated the description