xyz-tools/gcode-preview #231: Move the drag and drop code

sophiedeziel, 2024-10-16:
Should we un-deprecate the drag and drop option?

The `_enableDropHandler()` method could be moved outside of `WebGLPreview`.

remcoder, 2024-10-17:
I'd be for undeprecating DnD. We could probably use something like a a dom-utils module or canvas wrapper class to organize things like DnD and resizing.

remcoder, 2025-01-14:
petition to rename 'allowDragNDrop' to `dnd`. So you can just set the option `dnd : true`

sophiedeziel, 2025-01-23:
I vote `droppable`!

I'm not a fan of using acronyms in APIs, unless documentation is using the long form. The thing is, even when an acronym is widely known, someone may need to google it to know what it means. `I18n` is easy to find the meaning for, but `dnd` could mean a lot of things.

PR #284 (2025-05-25): Undeprecate dnd and move to dom-utils
fixes #231 

I totally forgot we have drag n drop in 2 places now... Not sure if this is the right moment to deduplicate it but I want to give it a try.