xyz-tools/gcode-preview #272: Move demo app to separate repo

remcoder, 2025-01-12:


remcoder, 2025-01-14:
@sophiedeziel how do you feel about this?

remcoder, 2025-01-17:
Ok I have to agree with what you say.

I guess what I hadn't mentioned is my reasoning for the split.

The demo has 2 roles, one being a showcase for the lib and other being a deployed application in and of itself. 

Lately, with the work on the alpha it felt like I was polluting the main repo whenever I was  making changes to the demo. And it it would help me with getting a sense of overview in a more focused repo.

But yeah we need to have some host app to dev the lib with, without needing a complicated setup. 

If the demo is in a separate repo, we can still use it for dev purposes but it will definitely require a more effort to setup.  

That setup is the thing that needs to be validated imo. And not just by one person. 

How about we let this marinate for a while 😉

sophiedeziel, 2025-01-23:
I've felt the same!

We've identified a need for something we can absolutely fill with tons of devs concerns, flexibility and setup. And a need for keeping the demo what it is: a demo.

What if we extract the demo code in a new repo without removing it from here? Then, rename it in this repo as `dev-tool` or something like that. The actual demo would be remain clean, and we'll have a starting point for a development wrapper app. Both will evolve their own way.

remcoder, 2025-07-25:
closing this for now