xyz-tools/gcode-preview #278: Travel lines are clipped

remcoder, 2025-01-14:
Often there are travel moves above the topmost extrusion layer. Due to the clipping planes, of which the max is initialized at the topmost extrusion layer, those travel lines are clipped. 

```
      x--------x  Travel move (INVISIBLE due to clipping)
----------------  clipping plane
================  Layer n (topmost extrusion layer)
================  Layer n-1
================  Layer n-2

```

sophiedeziel, 2025-01-27:
It's an interesting challenge to solve. 

The clipping mask is needed to hide travel lines of layers above the upper limit, but we want lines that *start* on a visible layer to show in full. 

A solution could be to detect such travel paths that should be visible and have some parts hidden and render them independently without the clipping mask. I don't remember if travel lines are rendered in batches like the extrusion paths. If that's the case, we'd have to exclude them from the batch. 

I think fixing this would be most helpful when z-hop is enabled on the slicer. Travels would systematically be "lost" on the last layer. 

Maybe clipping was a good idea for performance and ease of implementation, maybe we need to reconsider the approach?

I'm also thinking about an hybrid approach: clipping could be used for a 1st render that's fast, followed by an actual render of the subset of visible layers. We'd get the best of both worlds.

sophiedeziel, 2025-01-27:
> I suggest that at least we extend the parser to read and store this info. Agree?

Yes, 100%. We can start with a basic implementation first. We talked about this many times in the past, gcode metatada in the form of comments is a bit different from slicer to slicer and we could have dictionaries and stuff. We don't absolutely need all of this and to over-engineer it as long as we can iterate on it. 

>And I also suggest to first fix the most basic version of the problem: initially no min/max should be set and we should see the whole thing, with travel lines way above the print. Right?

Oh, I did not realize the clipping mask was there the whole time! Yes, 100%! This case is the most important and the easiest to fix.