xyz-tools/gcode-preview #285: THREE.LineSegmentsGeometry.computeBoundingSphere(): Computed radius is NaN. The instanced position data is likely to have NaN values.

Christian-Ackermann, 2025-01-28:
Hello,

Since last night, I have been receiving an error message when rendering GCode files, like the one I have attached to this ticket (I was made renaming the file from .gcode to .txt because of GitHub). Since the "[Svelte example](https://github.com/xyz-tools/gcode-preview-svelte)" loads without any issues, I suspect the problem lies on my end. I have already tried resetting the code via Git to an earlier working state, but the error persists. Currently, I am using the GCode-Preview version "3.0.0-alpha.2," but even the release version 2.18.0 results in the same error. Can you suggest which direction I should take for troubleshooting? To clarify, I have attached the code of my Svelte component.

Thanks, Christian

My svelte component:

```
<script lang="ts">
    import {onMount} from 'svelte';
    import * as GCodePreview from 'gcode-preview';

    import translation from '../../../store/translation';
    import type {GCodeFile} from '../../../types/components';

    import {GCode as GCodeUtils} from '../../../utilities/gcode';

    interface IProperties {
        file?: GCodeFile | null
        invertedZAxis?: boolean
    }

    let {
        file,
        invertedZAxis = false
    }: IProperties = $props();

    let canvas: HTMLCanvasElement;

    //region hooks

    onMount(() => {
        // is dark mode active?
        const html = document.getElementsByTagName('html')[0];
        const bgPreviewDark = '#1e293b';
        const bgPreviewLight = '#d1d5db';

        if (html) {
            // create the observer
            const observer = new MutationObserver((mutations) => {
                for (const mutation of mutations) {
                    // check if the class attribute has changed
                    if (mutation.type === 'attributes' && mutation.attributeName === 'class') {
                        preview.backgroundColor = html.classList.contains('dark') ? bgPreviewDark : bgPreviewLight;
                    }
                }
            });

            // start the observer
            observer.observe(html, {attributes: true, attributeFilter: ['class']});
        }

        // initialize GCode canvas
        const preview = GCodePreview.init({
            canvas,
            lineWidth: 3,
            renderTravel: true,
            renderExtrusion: false,
            backgroundColor: html && html.classList.contains('dark') ? bgPreviewDark : bgPreviewLight
        });

        // everytime the file object changes, the view must be updated
        $effect(() => {
            if (file && file.raw && file.raw.length > 0) {
                preview.clear(); // remove existing items
                preview.processGCode(prepareGCode(file.raw, invertedZAxis)); // draw new code
            }
        });
    });

    //endregion

    /**
     * Prepare the GCode for the preview plugin by removing the line counter and inverting the z axis if necessary.
     *
     * @param lines The lines of gcode as a string-array
     * @param invert If true, all z-axis values are with a minus sign.
     * @return       The modified gcode.
     */
    const prepareGCode = (lines: string[], invert: boolean): string[] => {
        // remove numbers / line counting from the code file
        let prepared: string[] = GCodeUtils.removeLineCounter(lines);

        // change sign of z axis
        return GCodeUtils.setZSign(prepared, invert);
    }

</script>
<canvas bind:this={ canvas } class="rounded-lg w-full"></canvas>

<p class="text-xs font-medium leading-relaxed text-gray-500 dark:text-gray-400/75">
    {#if file}
        { translation.translate('page.file.fusion.preview.gcode.selected.file.title') } { file?.name }

    {:else}
        &nbsp;

    {/if}
</p>

<style>
    canvas {
        cursor: grab;
    }
</style>

[1005-Kontur-Holz-6mm-Z2.txt](https://github.com/user-attachments/files/18568787/1005-Kontur-Holz-6mm-Z2.txt)

```

remcoder, 2025-01-28:
Hi Christian, 

You put the error message in the title but GitHub broke it off. Can you give us the full error message, including the stack trace?

Btw, have you checked with different gcode inputs? The error sounds like something that happens with zero-length lines.

And do you have any idea what changed in the meantime? Like maybe you did a new npm install and got a later version of Threejs? 

Tüß!

Remco

Christian-Ackermann, 2025-01-28:
Hi Remco,

thank you so much for your quick support and I'm sorry for writing the error message in the title. :(

Here are the messages from the console:
```
06:57:42.106 client:742 [vite] connecting...
06:57:42.110 client:861 [vite] connected.
06:57:42.253 VisualizeGCode.svelte:46 Using THREE r171
06:57:42.253 VisualizeGCode.svelte:46 opts {canvas: canvas.rounded-lg.w-full.s-MvUgGKu2JVuE, lineWidth: 3, renderTravel: true, renderExtrusion: false, backgroundColor: '#1e293b'}
09:00:06.762 client:461 [Violation] 'setInterval' handler took 87ms
09:19:33.290 VisualizeGCode.svelte:58 THREE.LineSegmentsGeometry.computeBoundingSphere(): Computed radius is NaN. The instanced position data is likely to have NaN values. lt {isBufferGeometry: true, uuid: 'aa596f1e-f2d2-451b-806c-b90585cedc5c', name: '', type: 'LineSegmentsGeometry', index: Uint16BufferAttribute, …}attributes: {position: Float32BufferAttribute, uv: Float32BufferAttribute, instanceStart: _InterleavedBufferAttribute, instanceEnd: _InterleavedBufferAttribute}boundingBox: Box3 {isBox3: true, min: _Vector3, max: _Vector3}boundingSphere: Sphere {isSphere: true, center: _Vector3, radius: NaN}drawRange: {start: 0, count: Infinity}groups: []index: Uint16BufferAttribute {isBufferAttribute: true, name: '', array: Uint16Array(18), itemSize: 1, count: 18, …}indirect: nullinstanceCount: 859isBufferGeometry: trueisInstancedBufferGeometry: trueisLineSegmentsGeometry: truemorphAttributes: {}morphTargetsRelative: falsename: ""type: "LineSegmentsGeometry"userData: {}uuid: "aa596f1e-f2d2-451b-806c-b90585cedc5c"_listeners: {dispose: Array(1)}_maxInstanceCount: 859id: 3[[Prototype]]: InstancedBufferGeometry
computeBoundingSphere @ gcode-preview.js?v=9dfe3cb7:27360
setPositions @ gcode-preview.js?v=9dfe3cb7:27326
renderPathsAsLines @ gcode-preview.js?v=9dfe3cb7:28020
renderPaths @ gcode-preview.js?v=9dfe3cb7:28009
render @ gcode-preview.js?v=9dfe3cb7:27960
processGCode @ gcode-preview.js?v=9dfe3cb7:27941
$effect @ VisualizeGCode.svelte:58
update_reaction @ chunk-6B2HMSQ2.js?v=9dfe3cb7:1308
update_effect @ chunk-6B2HMSQ2.js?v=9dfe3cb7:1399
flush_queued_effects @ chunk-6B2HMSQ2.js?v=9dfe3cb7:1483
flush_queued_root_effects @ chunk-6B2HMSQ2.js?v=9dfe3cb7:1469
process_deferred @ chunk-6B2HMSQ2.js?v=9dfe3cb7:1505
```

Regarding your question about other GCode files: I have tried many different files, including the ones I’m 100% sure worked before. I might rule out that I have a different version of Three installed, since I restored the "node_modules" folder from a backup and even if I copy the "node_modules" folder from "your svelte-example - project" (which is working) into my project, the error still appears. :(

I'm really sorry I can’t find the error myself. :(

remcoder, 2025-01-28:
No worries.. the error says there are NaN values somewhere. Which might be caused by a division by zero or some other calculation on the numbers. I notices you do some preprocessing on the gcode? Have you checked that the preprocessing works? Maybe test with a gcode file that has a single `G0 X0` command and go from there?

You GCodeUtils seem to be a likely source of bugs. Have you tried using gcode-preview without the preprocessing?

remcoder, 2025-01-28:
You could also try chatGPT btw. I asked it and it said you need to validate the input data.... which could work.

https://chatgpt.com/share/6798dd4e-f650-8010-a88d-322de895af61

Christian-Ackermann, 2025-01-28:
Hi,

after a lot of "uninstall" and "install" gcode-preview I was able to get the version 2.18.0 working. 
I suspect that it may be due to the version of Three. In the version that works, Three 159 is used, but in the alpha of gcode-preview 171. My “ GCodeUtils” should not be the cause, because even without them when using the GCOdes from the readme “const gcode = 'G0 X0 Y0 Z0.2\nG1 X42 Y42 E10';” the error also appears.

remcoder, 2025-01-28:
ah good find!

If you can post a reproduction case we can see if there is a bug in gcode-preview (in combination with threejs).

remcoder, 2025-01-29:
I can reproduce the error with this snippet:

```
      preview = new GCodePreview.init({
        canvas,
        lineWidth: 1,
        renderTravel: true,
        renderExtrusion: true,
        renderTubes: true,
        buildVolume: {
          x: 50,
          y: 50,
          z: 0
        }
      });

      preview.clear();
      preview.processGCode('G0 X0 Y0 Z0.2\nG1 X42 Y42 E10');
```

When it is rendering lines (renderTravel: true) it will produce the error.

Same thing for rendering extrusion as lines:

```
        renderTravel: false,
        renderExtrusion: true,
        renderTubes: false,
```

So there is something buggy in the line rendering. The weird thing is that the bug doesn't show up in the demo, which also uses Threejs r172.

Flagging this as a bug which needs to be fixed.

remcoder, 2025-01-29:
Hi, thanks for posting the repro!

I see now that error results from `lineHeight` not being set. This is a bug in gcode-preview. It should fallback to 0.2mm if no lineHeight is given. I will fix this and publish a new alpha version.

Thanks again for reporting!

remcoder, 2025-01-29:
FWIW, @Christian-Ackermann I just verified that the stable version does not have this bug. 
https://codepen.io/remcoder/pen/GgKzPNX

Christian-Ackermann, 2025-01-29:
Awesome!!! Thank you so much for your great support and helpful plugin!!!!

PR #286 (2025-01-29): Set a default value for lineHeight
fixes #285