xyz-tools/gcode-preview #307: Build volume dimensions not updated

remcoder, 2025-07-21:
..when switching presets

sophiedeziel, 2026-09-10:
I just checked this one on https://gcode-preview.web.app/

I think it is fixed, unless I am misunderstanding what the bug is. Can you check that it behaves the intended way and not what you observed?

remcoder, 2026-10-25:
Could not reproduce on `develop` @ 7b6fcfc (3.0.0-alpha.7), demo served with `npm run build && npm run live-server` in a Chromium-based browser.

Steps: switch through every preset in the demo's preset dropdown and compare `_preview.sceneManager.buildVolume` (and the Build volume tab's x/y/z/small-grid inputs) against the preset's `buildVolume` in `demo/js/presets.js`, merged over `demo/js/default-settings.js`.

All ten presets matched, both in the scene manager and in the settings inputs — e.g. benchy 180/180/0, arcs 130/150/0 + small grid, mach3 10/10/0, easel 300/180/0 — in forward order and switching back and forth.

Limitations: this only covers switching presets from the dropdown with no manual edits in between. If the original report was about editing the x/y/z fields by hand and *then* switching presets, or about a build volume that renders at the old size while the numbers are right, the repro steps for that would help. Leaving the issue open; no `not reproducible` label exists in this repo, so none was applied.